NeuronRobotics/nrjavaserial #29: Binary release lacks OSGi meta-data

kaikreuzer, 2015-03-03:
The binary jar at https://github.com/NeuronRobotics/nrjavaserial/releases/download/3.9.3/nrjavaserial-3.9.3.jar is no valid OSGi bundle. This is a shame, because the build scripts contain all required information here https://github.com/NeuronRobotics/nrjavaserial/blob/master/nrjavaserial/pom.xml#L75.

Is there a chance to build the official release with these entries in the MANIFEST.MF? This would make it directly usable in any OSGi context.

madhephaestus, 2015-03-03:
Hey, this is a neat idea. I would be interested in merging a pull request from anyone who knows how to add this information. I have never worked with OSGi, so I will not have the spare cycles to do this myself.

kaikreuzer, 2015-03-03:
I think no PR is required - it is a matter of how you do the build. I assume you are doing it through Ant.
If done by Maven by ```mvn clean install``` you will find the correct artifact under ```nrjavaserial/target/nrjavaserial-3.9.3.jar```.

madhephaestus, 2015-03-03:
the builders are run on a remote server automatically, so if it needs to be in the jar, it needs to be in the Ant build. Again, i am unfamiliar with OSGi, so if you can go in and edit https://github.com/NeuronRobotics/nrjavaserial/blob/master/nrjavaserial/build.xml  with the changes needed to build the correct artifacts, i can merge it in with the main sources.

kaikreuzer, 2015-03-03:
This is not about OSGi at all, this is about running a build through Maven vs. Ant. I guess you would have to update the build.xml to call Maven for compiling and packaging of the jar. But I am no Ant expert...
But as you have a pom.xml in the repo, I assume that this has been used in the past. Do you have any memories when and how this was done?

madhephaestus, 2015-03-03:
oh god, so long ago... 

The pom.xml is fairly out of date, it refers back to the google code repo from over 2 years ago, so i do not think anyone has used it since. I dont know if you noticed, but i would love some help maintaining this project, since some of the components are bits i just forked and have not been maintained correctly. I built the auto-jni-extractor engine and maintain a .jar with up to date compiled os binaries. The rest of the codebase and build scripts were inherited and very out of sync with the original RXTX project.

MrDOS, 2015-03-03:
Before the school year started, I was trying to work towards repeatable cross-platform builds made on a single build server (see #19). But, I never got as far as looking into jar packaging – when I stopped, I was still working on compiling the native libraries.

I think I was hoping to eventually discard all the Ant stuff and accomplish all aspects of the build (including kicking off native builds) through Maven as that seems to be the convention for modern Java projects, but I never got that far.

moberhuber, 2015-03-04:
I would also be very interested, especially if it would lead to nrjavaserial being usable unchanged in Eclipse !

Thanks,
Martin
--
Martin Oberhuber, SMTS / Product Owner – Development Tools, Wind River
direct +43.662.457915.85  fax +43.662.457915.6

MrDOS, 2015-11-18:
@kaikreuzer Here's an example of the sort of manifest #42 creates. Does this satisfy your use-case?

    Manifest-Version: 1.0
    Export-Package: gnu.io;version="3.11.0",gnu.io.rfc2217;version="3.11.0
     ";uses:="gnu.io,org.apache.commons.net.telnet"
    Bundle-SymbolicName: com.neuronrobotics.nrjavaserial
    Bundle-Version: 3.11.0
    Bundle-Name: nrjavaserial
    Bundle-ManifestVersion: 2
    Bnd-LastModified: 1447881535000
    Import-Package: org.apache.commons.net.telnet;version="[3.3,4)"
    Require-Capability: osgi.ee;filter:="(&(osgi.ee=JavaSE)(version=1.6))"
    Created-By: 1.8.0_45 (Oracle Corporation)
    Tool: Bnd-2.4.0.201411031534

kaikreuzer, 2015-11-19:
@MrDOS, this looks great, yes!
The only concern I have is the line
`Import-Package: org.apache.commons.net.telnet;version="[3.3,4)"`
Is this really a hard dependency that the code needs in order to run? It probably is related to rfc2217, which is only relevant for a few users.
Is there any chance to have this import declared optional?

andyrozman, 2015-11-19:
It is required for compiling. If you don't need it, you don't need it to supply it with your end product (it is what I do with my product).

kaikreuzer, 2015-11-19:
> you don't need it to supply it

Well, if you are using it within OSGi (what this issue is all about), then you HAVE to supply this dependency as otherwise the com.neuronrobotics.nrjavaserial bundle won't be resolved :-(

MrDOS, 2015-11-19:
What's particularly weird is that if you look inside our JAR, we seem to be bundling `org.apache.commons.net.telnet`. So I think it would be a dependency if we were only using it as part of the RFC 2217 stuff, but because we build a fat JAR, we definitely don't need to require it as part of the OSGi manifest. I suspect the solution is to modify the scope of the dependency in `build.gradle`; I'll take a closer look.

MrDOS, 2015-11-19:
I've made some further changes. Comments?

Manifest-Version: 1.0
    Implementation-Title: nrjavaserial
    Bundle-SymbolicName: com.neuronrobotics.nrjavaserial
    Implementation-Version: 3.11.0
    Bundle-ManifestVersion: 2
    Bnd-LastModified: 1447958686000
    Specification-Vendor: Neuron Robotics Cooperative
    Specification-Title: nrjavaserial
    Require-Capability: osgi.ee;filter:="(&(osgi.ee=JavaSE)(version=1.6))"
    Tool: Bnd-2.4.0.201411031534
    Implementation-Vendor: Neuron Robotics Cooperative
    Export-Package: gnu.io;version="3.11.0",gnu.io.rfc2217;version="3.11.0
     ";uses:="gnu.io"
    Bundle-Name: nrjavaserial
    Bundle-Version: 3.11.0
    Created-By: Neuron Robotics Cooperative
    Specification-Version: 3.11.0

kaikreuzer, 2015-11-19:
Looks better to me.
Although your comment about the fat jar worries me. I am still on 3.9.3 and this exports only gnu.io and this is all I am interested in. I wouldn't want this jar to be blown up with external libs for some optional functionality. I think it would be better to have a slim jar as a download (and maybe a fat one as a second option).

closed 2015-11-19 by MrDOS

MrDOS, 2015-11-19:
I do see where you're coming from wrt library size. The packaging is convenient but blows up the size of the JAR from a little under 600KB to 1.1MB. Is half a MB of overhead a big concern? Not for large applications like the one I'm using NRJS in, but maybe for smaller projects. We're certainly not using all of `commons-net` even though we include all its classes; it may be possible to include only a subset of the library in our JAR.

If #43 satisfies yours and @moberhuber's OSGi requirements, I think we can close this issue and carry on the fat JAR slimming discussion in another issue.

(Anyone, however, who still has beef over the OSGi implementation should feel free to reopen and comment on this issue or open a new issue.)

kaikreuzer, 2015-11-19:
> Is half a MB of overhead a big concern?

Yes, it definitely is if you are running on embedded devices, such as a RaspPi.
I have created https://github.com/NeuronRobotics/nrjavaserial/issues/44 for the follow-up discussions.

PR #43 (2015-11-19): Exclude Commons packages from the OSGi imports
We don't want to list Apache Commons Net as an OSGi dependency as we build it into the JAR. This pull request is a follow-up to #42 and closes #29.