NeuronRobotics/nrjavaserial #33: Serial port I/O best pratices

jairocgr, 2015-07-28:
Hi,

***TLDR;*** I'm looking for best pratice/high performance for serial port reading and writing.

## Background

I'm quite new on Java I/O so i took a bunch of readings about the subject, but all information that i found was talking about networking and filesystem reading and writing.

I work with agriculture software, therefore i need to communicate with all sort of sensors(scales, rfid readers, termometers, anemometer, etc).

On my current project, the software must be able to handle concurrents weighing from a bunch of scales dispersed all over the farm. The weighing data will come via radio and will be forwarded for my software via USB/Serial COM port.

## What i am currently doing

Currently, i'm adding a `SerialPortEventListener` who relies on the internal `MonitorThread` from ***nrjavaserial*** library. My listener is acting as a dispatcher, it handles serial port events, perform i/o, wraps the inputs and dispatch them for another threads to handle it.

When the listener recieve the `SerialPortEvent.DATA_AVAILABLE`, it performs a byte-by-byte reading(using directly the the nrjavaserial's `InputStream`) until it find a line-feed char(`'\n'`). Just after the reading, it wraps the string that were read on a new event and post to a [Guava's EventBus](https://github.com/google/guava/wiki/EventBusExplained).

## What i'm exactly looking for

In order to design a proper solution, i have to know more about Serial port I/O. I want be able to trade-off about byte-by-byte/buffered reading, buffer size, multi-thread strategy and so on...

I want to build reliable and responsive system with a modest memory footprint.

Maybe you guys can help me ;)

madhephaestus, 2015-07-28:
The internal eventing thread is a simple way to access the serial port
asynchronously, but if performance is your goal, then setting up your own
polling thread is going to give you the best throughput.

I usually access the serial port using the streeam interface as seen here:

https://github.com/NeuronRobotics/java-bowler/blob/development/src/main/java/com/neuronrobotics/sdk/serial/SerialConnection.java#L150

And internally i poll the port for availible bytes:

https://github.com/NeuronRobotics/java-bowler/blob/development/src/main/java/com/neuronrobotics/sdk/common/BowlerAbstractConnection.java#L1246

I implemented my own ByteList in the java-bowler library to use a static
circular buffer that only resizes if the circular buffer overflows. THis
allows for a very high speed buffering, but tollerence of unknown  buffer
size conditions.

jairocgr, 2015-07-28:
Thx @madhephaestus, i'll study your implementation.

jairocgr, 2015-08-01:
@madhephaestus  i studied your implementation and i also did more digging on the Serial Impl.c too.

I'm thinking to abandon the SerialPortEventListener/MonitorThread approach and start polling the stream like you said.

One thing that i didn't figure out is if has some drawbacks in choosing the byte-by-byte reading or the avaliable/buffer reading.

andyrozman, 2015-11-19:
It depends on device... I am working with medical devices, and my latest has serial problems with byte-by-byte reading... After some time communication stops. Reading of buffer, helps eliminate this problem. So it really depends on device... If you know that device has special buffer length, it would be probably best to use that, and also it will be little faster.

madhephaestus, 2015-11-23:
Hmm, i had an issue with this too and found that it was a firmware issue on the microcontroller. You wouldnt happen to be using a Pic32 with the USB stack device side?

Also, just for my own interest, what medical device? I work in the WPI AIM lab http://aimlab.wpi.edu/ on surgical robots using java-bowler and this library, so i would be interested to know what other divices its used in.

andyrozman, 2015-11-23:
My project is Diabetes Management Software (GNU Gluco Control), so I use it with diabetes devices (glucose meters, pumps and CGMS devices)... I had problem with Minimed Pump, while using very old interface (there are several options how to connect to device, and oldest one had problems). While using this interface there is a lot of data that needs to be  read and my byte-by-byte implementation had quite few problems with that. It started working better after I started reading bigger packets.

closed 2020-05-06 by madhephaestus