NeuronRobotics/nrjavaserial #36: Constant CPU usage by gnu.io.RXTXPort.eventLoop

TheNetStriker, 2015-11-10:
I've just tested this library on my BeagleBone Black and I noticed that the gnu.io.RXTXPort.eventLoop is constantly using about 15% percent of the cpu. (Even if there is no activity on the serial port) I found an older issue where this problem was discussed before:
https://github.com/NeuronRobotics/nrjavaserial/issues/22

The solution that is described there does not work for me because I'am already using the serialEvent to receive data from the serial port. How can this problem be solved?

madhephaestus, 2015-11-11:
A solution was merged in December, and current version is 3.10.2
What version are you using?

TheNetStriker, 2015-11-11:
I've just downloaded 3.10.2 from here: http://mvnrepository.com/artifact/com.neuronrobotics/nrjavaserial/3.10.2

But this version still constantly uses 15% cpu. I've also tested this on my mac and there is the cpu load constantly 6%. (On a Xeon cpu) Do I also have to configure something to reduce the cpu load?

TheNetStriker, 2015-11-17:
I am now using two serial ports and as expected the cpu usage is now constantly about 30%. Is there any hope to fix this?

madhephaestus, 2015-11-17:
Can you run VisualVM and tell me which function is taking so much time on your system? If eventLoop is the time consuming one, then we can deal with that, but it could theoretically be any function in the eventing path. Note that the JNI event loop sends an event, triggering events in Java, meaning java performance issues can back up into the JNI execution, so be careful about execution times of events. 

Can you get the JNI compilation environment running so if we can work on a solution together?

The monitor thread simply kicks off the event loop function in the JNI. 
https://github.com/NeuronRobotics/nrjavaserial/blob/master/src/main/java/gnu/io/RXTXPort.java#L1611

Here is where the JNI picks up:
https://github.com/NeuronRobotics/nrjavaserial/blob/master/src/main/c/src/SerialImp.c#L4200

You can add delays into the polling loop but suddenly run the actual risk of missing bytes coming in. This will lower the processor load but add unreliability. Not a compromise most applications are willing to sacrifice. 

If you will notice there is a sleep command for each loop: 
https://github.com/NeuronRobotics/nrjavaserial/blob/master/src/main/c/src/SerialImp.c#L4253

set to 1000 us (aka 1 ms). It would be possible to change the JNI API to add a parameter to eventLoop so you could at runtime set how much time is waits between events, but to distribute something like that i would need all of the compiled JNI architectures updated with this new JNI API. Before i begin such a massive update i would want confirmation that this issue can be resolved on the target system first, so you setting up the build here is critical.

TheNetStriker, 2015-11-17:
I've attached VisualVM to my (openhab) java instance. Here is the output of the profiler:

[image]

And here are the Threads from the thread dump:

"Thread-51" - Thread t@119
   java.lang.Thread.State: RUNNABLE
	at gnu.io.RXTXPort.eventLoop(Native Method)
	at gnu.io.RXTXPort$MonitorThread.run(RXTXPort.java:1611)

"Thread-31" - Thread t@80
   java.lang.Thread.State: RUNNABLE
	at gnu.io.RXTXPort.eventLoop(Native Method)
	at gnu.io.RXTXPort$MonitorThread.run(RXTXPort.java:1611)

I've already tried to setup a JNI compilation environment but I wasn't able to get the compilation to work. (Under Ubuntu) Is there somewhere a manual on how to setup the compilation environment and start the compilation?

madhephaestus, 2015-11-17:
First try the make command found:

https://github.com/NeuronRobotics/nrjavaserial#building-native-code

TheNetStriker, 2015-11-17:
My device runs on an armV7 cpu. When I execute the "make arm" command I get the following error:

Native output directories seem to exist.
arm-linux-gnueabi-gcc -I/usr/lib/jvm/java-7-openjdk-amd64/include/ -I/usr/lib/jvm/java-7-openjdk-i386/include/ -I"./include" -I"./include/target" -I/usr/lib/jvm/java-6-openjdk-amd64/include/ -I/usr/lib/jvm/java-6-openjdk-armhf/include/ -I/usr/lib/jvm/java-6-openjdk-arm/include/ -I/usr/lib/jvm/java-6-openjdk-i386/include/ -O3 -Wall -c -fmessage-length=0 -fPIC -MMD -MMD -MP --with-arch=armv7  src/fixup.c -o build/fixup.o
arm-linux-gnueabi-gcc: error: unrecognized command line option ‘--with-arch=armv7’
make[1]: *** [arm7] Error 1
make[1]: Leaving directory `/home/mad/Dokumente/GitHub/nrjavaserial/nrjavaserial/src/main/c'
make: *** [arm] Error 2

How can I solve this problem?

madhephaestus, 2015-11-18:
Ah, i think onthe native compile you might need to screw with the compile command. It is assuming it is on x86 and cross compiling to armv7, not native compiling. Look at the x86 compile instructions, and maybe make a new compile tag for compiling arm *on arm*, which the old crufty makefile does not yet have a target for.

TheNetStriker, 2015-11-18:
I finally got the build working by only compiling using "make arm7HF" in the src/main/c directory. With this I was able to compile libNRJavaSerialv6_HF.so for my beaglebone. I also had to add the paths to my oracle java installation to the makefile:

-I/usr/lib/jvm/java-7-oracle/include -I/usr/lib/jvm/java-7-oracle/include/linux

I now changed the "usleep(1000)" to "usleep(100000)", but that just made the cpu load worse. What else would you like me to test?

madhephaestus, 2015-11-18:
And if you make it smaller...?

MrDOS, 2015-11-18:
I'm surprised changing [line 4253](https://github.com/NeuronRobotics/nrjavaserial/blob/master/src/main/c/src/SerialImp.c#L4253) changed anything, as it seems to be in a Windows-only `#ifdef`. Or did you change [line 3136](https://github.com/NeuronRobotics/nrjavaserial/blob/master/src/main/c/src/SerialImp.c#L3136)?

TheNetStriker, 2015-11-19:
Ok, maybe it was just the additional CPU usage that VisualVM produced on my device. I've now compiled one version with a lower timeout at line 3136 (10) and one with a higher timeout (100000) but the CPU usage was about the same on the serial port threads. Any other suggestions what I could try to modify?

madhephaestus, 2015-11-19:
I am struggling with how it could have made the CPU usage worse to make it bigger. usleep is supposed to be signaling the kernel not to schedule this process for processing during that time. If you make the wait super large, will it taper off usage? maybe its getting caught in transitions from active and not? 

What your describing sounds like usleep is not working according to its spec, as if it was a busy wait. What happens when you comment it out entirely?

MrDOS, 2015-11-19:
I'm curious about the timeout behaviour of the [`select(2)`](http://linux.die.net/man/2/select) call on [line 4230](https://github.com/NeuronRobotics/nrjavaserial/blob/master/src/main/c/src/SerialImp.c#L4230). The function takes a timeout as its last parameter in a loop, and on Linux the struct fields are modified to reflect the unused portion of the timeout. The timeout is initialized to 1ms on [line 4149](https://github.com/NeuronRobotics/nrjavaserial/blob/master/src/main/c/src/SerialImp.c#L4149). So the `select` will sleep properly for the first 1ms of waiting and then effectively become a tight polling loop until a character is seen. (That's how I read it, at least.) I'd like to try inserting this right after the `select` call:

    if (eis.tv_sleep) {
        eis->tv_sleep.tv_sec = 0;
        eis->tv_sleep.tv_usec = 1000;
    }

I'm trying to get my environment together to test this, so don't feel compelled to try it yourself unless you're curious. I'm not wholly certain that this is the problem; good proof that it isn't the only area of difficulty is that OS X [doesn't seem to implement the same sort of decrementing behaviour on the timeout parameter](https://developer.apple.com/library/mac/documentation/Darwin/Reference/ManPages/man2/select.2.html) but I see the same CPU usage problem there.

TheNetStriker, 2015-11-19:
I've just seen that line 4253 ist after the #else statement, so it should be relevant too. I've now commented out lines 4253 and 3136 and the cpu load is still at the same level. I will later try the solution of MrDOS.

MrDOS, 2015-11-19:
It's after the `#else`, but the condition is `#ifndef WIN32`, not an `#ifdef WIN32`, so unless I'm missing something, it looks like the `#else` block is *only* executed on Windows.

TheNetStriker, 2015-11-19:
Yes i think you are right. I've tried to compile your solution but then I got compiler errors:

src/SerialImp.c: In function ‘read_byte_array’:
src/SerialImp.c:3097:4: warning: implicit declaration of function ‘$SELECT’ [-Wimplicit-function-declaration]
    ret = $SELECT( fd + 1, &rset, NULL, NULL, tvP );
    ^
src/SerialImp.c: In function ‘Java_gnu_io_RXTXPort_eventLoop’:
src/SerialImp.c:4232:5: error: used struct type value where scalar is required
 if (eis.tv_sleep) {
     ^
src/SerialImp.c:4233:8: error: invalid type argument of ‘->’ (have ‘struct event_info_struct’)
     eis->tv_sleep.tv_sec = 0;
        ^
src/SerialImp.c:4234:8: error: invalid type argument of ‘->’ (have ‘struct event_info_struct’)
     eis->tv_sleep.tv_usec = 1000;
        ^
src/SerialImp.c: In function ‘check_group_uucp’:
src/SerialImp.c:5777:12: warning: comparison between pointer and integer [enabled by default]
  if ( NULL == mkstemp(testLockAbsFileName) )
            ^
make: *** [arm6HF] Error 1

Can you give me an example how to solve this error?

MrDOS, 2015-11-19:
Increasing the timeout on lines 4148 and 4149 to 1 second/0 microseconds dramatically reduces CPU usage for me (~5% to 0.2%):

    eis->tv_sleep.tv_sec = 1;
    eis->tv_sleep.tv_usec = 0;

But before I make a pull request with this change, I want to confirm that this doesn't disturb the behaviour of event-based communication in any way. It *should* only add a little delay to removing a listener from a quiet serial line, but I want to be certain.

MrDOS, 2015-11-19:
The CPU usage I observe with the timeout cranked to 1s is on par with that seen with the workaround from #22 in place. As for behaviour, events still seem to be generated in a timely fashion – one would hope so, given the purpose of `select(2)`.

I'm going to pass this build over to a coworker to test for a day or two and if it all looks OK and @madhephaestus willing, we can probably release this as 3.12.0 as soon as I can coordinate another Linux build targeting glibc 2.5 – I think this is significant enough to warrant a release.

TheNetStriker, 2015-11-19:
I also tested this setting and can confirm that the cpu load is much lower. I also tested my serial based RFID door control and I didn't notice any additional delay to the original version.

MrDOS, 2015-11-19:
Out of curiosity, can you give me some percentages? I'm running on a recent i5 – sort of hard to get a perspective when you were seeing 15% of your CPU wasted to this.

Testing between two laptops, an event-based listener is able to consume data at pretty much exactly the expected baud rate (tested at 9600, 115200, and 230400 baud). I'm not sure what inferences can be taken from that, but it does mean that there's not much overhead in the picture. Cool.

TheNetStriker, 2015-11-19:
Before I had a constant cpu usage of 11% per serial port, now It's about 0.3 to 1% per serial port.

MrDOS, 2015-11-19:
Fantastic. To me, that sounds like some lower threshold based on the library doing other pieces of actual work.

Tomorrow, I'm going to try a range of other timeout values – try to get it a little lower than a second just to be reasonable. I suspect 100ms will work fine too.

Thanks for all your help tracking this down. I hope we can get a build out for you soon.

MrDOS, 2015-11-20:
100ms does work fine. I'm now in the process of rebuilding natives. Windows doesn't need changing and OS X is done; I'm just setting up some CentOS 5 VMs so I can build the 32- and 64-bit x86 Linux natives against glibc 2.5. Not sure how I'm going to compile for ARM yet; setting up a cross-compilation toolchain on CentOS sounds like pure torture but I don't want to land in the same glibc trap.

TheNetStriker, 2015-11-20:
The following setting also works for me:

eis->tv_sleep.tv_sec = 0;
eis->tv_sleep.tv_usec = 100000;

Same cpu load as with 1 sec.

commit ea5c0965 (2015-11-23): Reduce CPU usage of serial event generator.

There's no reason for the select(2) call to have a 1ms timeout. At that
point, it's basically a busy loop. A 100ms timeout greatly reduces CPU
usage while retaining fairly quick responsiveness to port closure.

Closes #36.
closed 2015-11-23 by MrDOS