NeuronRobotics/nrjavaserial #44: Provide a slim jar packaging besides the fat jar

kaikreuzer, 2015-11-19:
With 1.1MB version 3.11.0 doubled in size from 3.10.0 with 550KB.
As discussed in https://github.com/NeuronRobotics/nrjavaserial/issues/29#issuecomment-158179128, the reason is that it now includes an commons net as a dependency in a fat jar.
Specifically for OSGi use, it does not make any sense to package other OSGi bundles. Therefore it would be much desired to have a "pure" bundle as a download that does not contain this dependency.

MrDOS, 2015-11-23:
I'm really worried about the legal implications of including jvser (#28). But in terms of releases, I think we can accomplish two JARs pretty easily. One – the one that gets pushed to Maven – will let the typical packaging behaviour run its course. A second “fat” JAR definition will use the current packaging rules to include dependencies and strip dependency information from the OSGi manifest.

madhephaestus, 2015-11-23:
Yeah, this is one of those things Gradle makes possible with little effort. We can just add an additional call line to the uploadArchives section to build the thin jar and push it. If you pull it in through Maven, will the deps get dragged in too? I do not know enough about how mave deals with dependencies in the manafest that are not bundled in. If it deals with it intelligently we may be able to drop the fat jar entirely and let the slim jar+ Maven dependencies provide all the classes, though that may screw over people not using Maven...

Its worth investigating, but it looks like bundling both is an easy compromise?

MrDOS, 2015-11-23:
> If you pull it in through Maven, will the deps get dragged in too?

They're supposed to. Actually, I need to double-check that the auto-generated POM included in the fat JAR doesn't accidentally cause commons-net to be downloaded.

> though that may screw over people not using Maven...

I think this might be OK as long as we make it easy to find a list of dependencies (probably in the README, right after the Maven and Gradle inclusion instructions).

kaikreuzer, 2016-09-13:
Are there any news on the 3.12.0 release? I would love to upgrade to a more recent version, but as discussed in this issue, there should really be a slim jar again!

madhephaestus, 2016-09-13:
https://oss.sonatype.org/#nexus-search;gav~com.neuronrobotics~nrjavaserial~~~~kw,versionexpand is posted to Maven

kaikreuzer, 2016-09-13:
Hm, right, but that means that Github isn't really maintained very well anymore - it isn't mentioned on the [releases page](https://github.com/NeuronRobotics/nrjavaserial/releases) and the [3.12 milestone is still not closed](https://github.com/NeuronRobotics/nrjavaserial/milestone/1) as there are 3 open issues including this one here, which I think is pretty important (even from a legal perspective as mentioned in https://github.com/NeuronRobotics/nrjavaserial/milestone/1).

kaikreuzer, 2018-08-16:
Version 3.14.0 is shipped without dependencies again, so this issue can be closed, thanks!

closed 2018-08-16 by kaikreuzer