NeuronRobotics/nrjavaserial #45: What is the point of the “legacy” native libraries?

MrDOS, 2015-11-20:
The native resources include a whole bunch of libraries with “_legacy” in their names. What was the intent of these? Are they the original binaries imported from RXTX? The Makefile doesn't give any indication of how they're built, which makes me suspect they're more older versions than alternatives.

madhephaestus, 2015-11-20:
yeah, they were the linux 2.x kernel versions from the original rxtx project

MrDOS, 2015-11-20:
Ah, gotcha. I'll do some testing on older machines at some point with our newer libraries to see how they hold up. It might not be worth keeping them (but I'm loathe to toss them without knowing for sure that they're not necessary).

closed 2015-11-20 by MrDOS