NeuronRobotics/nrjavaserial #46: Cortex A8 libraries go unused

MrDOS, 2015-11-20:
[`OSUtil#isCortexA8()`](https://github.com/NeuronRobotics/nrjavaserial/blob/master/src/main/java/gnu/io/NativeResource.java#L231) is stubbed-out and not called from anywhere, so the A8 native libraries are a bit pointless at the moment. Could they just be included at the front of [the usual list of ARM libraries](https://github.com/NeuronRobotics/nrjavaserial/blob/master/src/main/java/gnu/io/NativeResource.java#L43)? Can anyone name a cheap/widespread piece of A8 hardware I can get my grubby mitts on to test this?

MrDOS, 2015-11-20:
Looks like I can use this as an excuse to splurge on a [BeagleBone Black](http://beagleboard.org/BLACK).

madhephaestus, 2020-05-06:
stale closing

closed 2020-05-06 by madhephaestus