NeuronRobotics/nrjavaserial #47: ARM library load order might be wrong

MrDOS, 2015-11-20:
I interpret the way the [list of ARM libraries](https://github.com/NeuronRobotics/nrjavaserial/blob/master/src/main/java/gnu/io/NativeResource.java#L43) is defined in code and [how they're iterated over and loaded](https://github.com/NeuronRobotics/nrjavaserial/blob/master/src/main/java/gnu/io/NativeResource.java#L50) as preferring the ARMv5 library, then ARMv6HF, ARMv6, ARMv7HF, and finally ARMv7. And if I understand how ARM backwards compatibility works, the ARMv5 library should work fine on later architectures. So I wonder if this code isn't mostly seeing success loading the ARMv5 library and not touching anything else.

@TheNetStriker, you were building ARMv7 – did you run into any troubles with this?

TheNetStriker, 2015-11-20:
On my device (BeagleBone) the libNRJavaSerialv5 cannot be loaded, then the libNRJavaSerialv6_HF loads and that library works. (I've got the two directories libNRJavaSerialv5_openhab_0 and libNRJavaSerialv6_HF_openhab_0 in my tmp directory)

MrDOS, 2015-11-20:
What did you have to do to get your ARMv7 build for #36 to work? Or were you just using your desktop in that case?

TheNetStriker, 2015-11-20:
I didn't get the build to work on my Ubuntu amd64 system, so I've built the library directly on the Beaglebone. But the BeagleBone only seams to be able to compile the HF builds.

mnlipp, 2016-01-18:
The load order is definitely wrong. I checked it with a Raspberry Pi 1 (armv6) and 2 (armv7). As things are currently, both load libNRJavaSerialv6_HF, because due to backward compatibility this also works fine on the armv7. (libNRJavaSerialv5 doesn't load, presumably because it uses soft float.)

I think the order should be

`private String[] armLibs = {"libNRJavaSerial_HF","libNRJavaSerial","libNRJavaSerialv6_HF","libNRJavaSerialv6","libNRJavaSerialv5"};`

This loads libNRJavaSerialv6_HF on the Raspberry Pi (1) and libNRJavaSerialv7_HR on the Raspberry Pi 2.

closed 2016-03-09 by madhephaestus