NeuronRobotics/nrjavaserial #50: GEt Gradle to build with Java 1.5 compatibility

MrDOS, 2015-11-23:
#21 asked us to compile for Java 1.5, and we've maintained that between then and now (I think). However, Gradle doesn't run on Java 1.5 – it has a minimum of 1.6. Before we build a new JAR for the 3.12.0 release, we need to figure out the configuration required to get Gradle to target 1.5 from a 1.6 compiler, or to get Gradle to invoke a 1.5 `javac` when running under 1.6.

andyrozman, 2015-11-23:
Wasn't somewhere said that we will build only for Java 6 and higher... 

I know that that was the main reason why I forked this project in the first place. 

My project is still on Java 5 because we want to support older devices (Macs seem to be the problem, on all other machines you can upgrade java as you wish)... Since we don't use any Java 6 or better functionalities we haven't had problem so far... Only "problem" is that I had to fork several  project so that I can build on 1.5...

andyrozman, 2015-11-23:
One unrelated question... Why did we switch to gradle? What was wrong with maven?

madhephaestus, 2015-11-23:
ease of use for new developers, self contained build environment (gradlew), easy module integration into other projects using git submodule/gradle subproject. 

It also offers the future development option of adding the C buildis into the single build script, even calling remote server compiles of the C artifacts can be integrated into gradle scripts.

MrDOS, 2015-11-23:
@andyrozman Out of curiosity, on which versions of OS X are you experiencing problems with Java 1.6?

andyrozman, 2015-11-23:
I am not sure which version is it excatly, but it's either 10.0.4 or 10.0.5 (I am not sure). I have an old Mac machine lying around.

MrDOS, 2015-11-23:
Holy smokes! And the natives I've been building still work with that? I'm amazed – I've been building against the 10.5 SDK.

andyrozman, 2015-11-23:
Actually. I might have 10.5, I am not sure... Only thing that I know for sure is, that there is no JDK 6 for my Mac... 

I haven't tested with my old Mac for a long while, so I don't know if it works or not... But I will have to do that soon... I just need to transport computer to my current home...

madhephaestus, 2016-01-11:
Have we come to a conclusion on if this is possible for the 3.12.0 release? If not, maybe postpone till the next release?

MrDOS, 2016-01-11:
I haven't had time to try yet. I'd still like to do so as 3.12.0 feels like a release focused on greater compatibility, but I understand if you'd like to push it back.

madhephaestus, 2016-01-11:
I pushed back the 3.12.0 target release date to the end of the month. No need to stress, I just want NrjavaSerial to be the best library we can release, and as up to date as we can keep it.

andyrozman, 2016-03-07:
Perhaps it would be enough to set target to 1.5... In one other project, we just set that target version of jar (and source) is 1.5 (maven-compiler-plugin) and that was enough. But that was in maven. Has gradle something similar?

andyrozman, 2016-03-23:
I found options, which we need. 
sourceCompatibility = '1.5'
targetCompatibility = '1.5'

This should take care of everything...

madhephaestus, 2020-05-06:
we are moving to java8 base compatibility

closed 2020-05-06 by madhephaestus