NeuronRobotics/nrjavaserial #51: Define compatibility goals

MrDOS, 2015-11-23:
We've had a number of issues asking us to support older environments (e.g., #16, #21). I'd like to think that we don't do too badly in this area, but what we're trying for remains a little wishy-washy. It would be nice to define more clearly which platforms we intend to support.

This is a little pie-in-the-sky right now and I've assigned it to me as the answer is partially dependent on the limits of automation (#19). However, if you (yes, you, the person reading this) have some specific platform requirement in mind – especially a non-x86 processor, or an old version of Java – please respond here so we have some idea of what's definitely desired by users of the library.

madhephaestus, 2015-11-23:
This might be a quick fix for this: 

https://discuss.gradle.org/t/how-do-iuse-the-jdk-1-4-compiler/6449

Gradle, while it requires 1.6 to run the build, it can target older JDK versions. 1.5 is very old at this point, but should not actually be a problem to target.

andyrozman, 2015-11-23:
I would be very thankful if we could still support Java 1.5. 

Currently I have a fork, in which I build nrjavaserial for 1.5, but I have to "manually" merge it with this repository...

MrDOS, 2015-11-25:
I'm not sure we need as many ARM targets as we currently have. [Debian's `armel` port](https://wiki.debian.org/ArmPorts) is compiled for ARMv4 and up, [Oracle Java SE Embedded](http://www.oracle.com/technetwork/java/embedded/embedded-se/downloads/index.html) only targets softfloat v5(/v6/v7) and hardfloat v6(/v7), and [Ubuntu](http://www.ubuntu.com/download/server/arm) aims for hardfloat v7 and 64-bit hardfloat v8. Assuming we nailed binary compatibility, I think we'd get pretty far with three distinctly different ARM natives: `armel` (v4+), `armhf` (hardfloat v6+), and `arm64` (hardfloat v8+). We wouldn't lose compatibility from where we are now, the build process would be more sane, and I doubt anyone would notice any performance difference.

madhephaestus, 2015-11-25:
This seems like a good comprise

madhephaestus, 2020-05-06:
stale closing

closed 2020-05-06 by madhephaestus