NeuronRobotics/nrjavaserial #53: read method behavior win/linux

gchauvet, 2015-12-02:
Hello,

I notice a different behavior between Windows (>= xp) and Linux : 
```java
int value = input.read();
while ((value & 0xFF) != ENQ) {
  System.out.println("Waiting....");
  value = input.read();
}
```
Under Windows, I can read message "Waiting" until ENQ code was reached. Under Linux, my code is stuck on the first input.read() call. In my real usecase, I use this loop to reach a lapse of time before throwing a TimeoutException.

Is it possible to have the same behavior between Windows & Linux ?

andyrozman, 2015-12-02:
It all depends on how you set your port. I get always -1 when there is no data available... (on linux). You can control if you have blocking or non-blocking. Google "rxtx java blocking"

gchauvet, 2015-12-02:
okay, this is possible :+1: 
```java
port.enableReceiveTimeout(200);
```
Solve my "issue"

Thank you !

closed 2015-12-02 by gchauvet

MrDOS, 2015-12-02:
See #27, and my [comment there](https://github.com/NeuronRobotics/nrjavaserial/issues/27#issuecomment-158470844). You're right, the out-of-box behaviour is unintuitive, but I suspect it's actually the Windows behaviour that's wrong. Either way, better documentation is needed. Thank you for the feedback.