NeuronRobotics/nrjavaserial #56: gradle build error on jcenter() method not found

guitarpicva, 2015-12-23:
I commented that line out of the build.gradle file and it appeared to work then, but I fear I have probably just ended up with an incomplete library.

`ab4mw@shack:~/src/nrjavaserial$ gradle --info build`
`Starting Build`
`Settings evaluated using empty settings script.`
`Projects loaded. Root project using build file '/home/ab4mw/src/nrjavaserial/build.gradle'.`
`Included projects: [root project 'nrjavaserial']`
`Evaluating root project 'nrjavaserial' using build file '/home/ab4mw/src/nrjavaserial/build.gradle'.`
` `
`FAILURE: Build failed with an exception.`
` `
`* Where:`
`Build file '/home/ab4mw/src/nrjavaserial/build.gradle' line: 41`
` `
`* What went wrong:`
`A problem occurred evaluating root project 'nrjavaserial'.`
`> Could not find method jcenter() for arguments [] on root project 'nrjavaserial'.`
` `
`* Try:`
`Run with --stacktrace option to get the stack trace. Run with --debug option to get more log output.`
` `
`BUILD FAILED`
` `
`Total time: 5.408 secs`

MrDOS, 2015-12-23:
`jcenter()` is a reference to the [Bintray Maven repository](https://bintray.com/bintray/jcenter) located at `http://jcenter.bintray.com/`. If the [`jcenter()` method](https://docs.gradle.org/current/javadoc/org/gradle/api/artifacts/dsl/RepositoryHandler.html#jcenter%28%29) isn't defined for you, it can be manually included like this:

```
repositories {
    mavenCentral()
    maven {
        url "https://jcenter.bintray.com"
    }
}
```

What version of Gradle are you running? I didn't realize the method was a recent addition.

As far as your concerns of an incomplete build, they're (happily) almost certainly unfounded: as the line in question just defines a repository, you'd get an error elsewhere in the process if a dependency couldn't be located because it existed in that repository. I believe the reference is currently included only for possible future developer convenience.

guitarpicva, 2015-12-23:
On Wed, Dec 23, 2015 at 9:29 AM, Samuel Coleman <notifications@github.com>
wrote:

> repositories { mavenCentral() maven { url "https://jcenter.bintray.com" }
> }

Samuel,

Thank you for the speedy reply.

My gradle is ( as installed by Debian Jessie repository ):

ab4mw@shack:~/src/nrjavaserial/build/libs$ gradle -v

------------------------------------------------------------
Gradle 1.5
------------------------------------------------------------

Gradle build time: Wednesday, October 8, 2014 8:21:39 AM UTC
Groovy: 1.8.6
Ant: Apache Ant(TM) version 1.9.4 compiled on October 7 2014
Ivy: non official version
JVM: 1.7.0_85 (Oracle Corporation 24.85-b03)
OS: Linux 3.16.0-4-amd64 amd64

--
Mitch Winkle
http://ewamjlu.blogspot.com
...How long will you go limping between two different opinions? If the LORD
is God, follow him...
1 Kings 18 ESV

guitarpicva, 2015-12-23:
The manual inclusion of the repository worked to eliminate my build error.

Many thanks,

Mitch

--
Mitch Winkle
http://ewamjlu.blogspot.com
...How long will you go limping between two different opinions? If the LORD
is God, follow him...
1 Kings 18 ESV

On Wed, Dec 23, 2015 at 9:40 AM, Mitch Winkle <mitchwinkle@gmail.com> wrote:

> On Wed, Dec 23, 2015 at 9:29 AM, Samuel Coleman <notifications@github.com>
> wrote:
>
>
>
> Samuel,
>
> Thank you for the speedy reply.
>
> My gradle is ( as installed by Debian Jessie repository ): [...]

MrDOS, 2015-12-23:
Looks like the `jcenter()` method was added to [incubation](https://docs.gradle.org/1.7/javadoc/org/gradle/api/Incubating.html) in [1.7](https://docs.gradle.org/1.7/javadoc/org/gradle/api/artifacts/dsl/RepositoryHandler.html#jcenter%28%29) and stablized in [1.10](https://docs.gradle.org/1.10/javadoc/org/gradle/api/artifacts/dsl/RepositoryHandler.html#jcenter%28%29).

If you'd like to be able to build the project sans modifications, I'd suggest using a newer Gradle version. On a Debian system, I'd pin the [Sid package](https://packages.debian.org/sid/gradle): the dependencies don't look too far-reaching so it should be pretty safe. If you're not comfortable doing that, it should also be pretty easy to get the [official binary]((https://gradle.org/gradle-download/)) up and running: put the unarchived `gradle-2.10` directory in `/opt` then symlink `/opt/gradle-2.10/bin/gradle` into `/usr/local/bin`. I haven't tried it but as long as Gradle will figure out where its home directory is by itself, it should work correctly.

commit 311ba7a9 (2016-01-11): close #56 reverting the jcentral to the more explicate version. 

There si no reason this should require a newer versioin of gradle, and
this change will not be removed in future releases.
closed 2016-01-11 by madhephaestus