NeuronRobotics/nrjavaserial #67: JVM crash when hardware in use disconnected

gchauvet, 2016-02-23:
Dear NRJavaSerial devs,

We have an issue with NRJavaSerial library who crash our JRE. This issue occurs on physical or virtual serial port (same behaviour).

```java
import gnu.io.NRSerialPort;
import gnu.io.SerialPortEvent;
import gnu.io.SerialPortEventListener;
import java.io.IOException;
import java.util.concurrent.Callable;
import java.util.concurrent.ExecutorService;
import java.util.concurrent.Executors;
import java.util.concurrent.Future;

public class MainApp {

    public static void main(String args[]) throws Exception {
        final String port = args[0];
        final int baudRate = Integer.valueOf(args[1]);
        final NRSerialPort serial = new NRSerialPort(port, baudRate);
        final ExecutorService executor = Executors.newSingleThreadExecutor();

        serial.connect();
        serial.getSerialPortInstance().notifyOnDSR(true);
        serial.getSerialPortInstance().notifyOnBreakInterrupt(true);
        serial.getSerialPortInstance().notifyOnFramingError(true);
        serial.getSerialPortInstance().notifyOnOverrunError(true);
        serial.getSerialPortInstance().notifyOnParityError(true);
        
        
        try {
            final Future<Void> task = executor.submit(new Callable<Void>() {
                @Override
                public Void call() throws Exception {
                    while (!Thread.currentThread().isInterrupted()) {
                        System.out.println(serial.getInputStream().read());
                    }
                    return null;
                }
            });
            
            serial.addEventListener(new SerialPortEventListener() {
                @Override
                public void serialEvent(SerialPortEvent ev) {
                    System.err.println("Receive code " + ev.getEventType());
                    serial.removeEventListener();
                    task.cancel(true);
                }
            });
            task.get();
        } catch (Exception ex) {
            throw new IOException(ex);
        } finally {
            serial.removeEventListener();
            serial.disconnect();
        }
    }

}
```

```batch
C:\temp\SerialNotifier>java -cp serialnotifier-1.0.0.jar;nrjavaserial-3.11.0.jar
 com.bimedia.utils.serialnotifier.MainApp COM8 9600
-1
-1
-1
-1
-1
-1
[... 223 lines cut ...]
0x76bf0000 - 0x76bfb000 	C:\Windows\system32\PSAPI.DLL

VM Arguments:
java_command: com.bimedia.utils.serialnotifier.MainApp COM8 9600
Launcher Type: SUN_STANDARD

Environment Variables:
PATH=C:\Windows\system32;C:\Windows;C:\Windows\System32\Wbem;C:\Windows\system32\WindowsPowerShell\v1.0;C:\httpd\Php\extras\wso2-wsf-php-bin-2.0.0-win32\wsf_c\lib\;C:\httpd\Php\extras\bindlib-cvs-vc8\lib\;C:\httpd\Php\extras\iconv-1.9.2.win32\bin\;C:\httpd\Php\extras\libxml2-2.7.6.win32\bin\;C:\httpd\Php\extras\openssl-0.9.8a.win32\bin\;c:\Program Files\Panda Security\WaAgent\Common;C:\Program Files\Skype\Phone\
USERNAME=Administrateur
OS=Windows_NT
PROCESSOR_IDENTIFIER=x86 Family 6 Model 28 Stepping 2, GenuineIntel

---------------  S Y S T E M  ---------------

OS: Windows XP Build 2600 Service Pack 3

CPU:total 2 (1 cores per cpu, 2 threads per core) family 6 model 28 stepping 2, cmov, cx8, fxsr, mmx, sse, sse2, sse3, ssse3, ht

Memory: 4k page, physical 2087020k(1352620k free), swap 2957980k(2166384k free)

vm_info: Java HotSpot(TM) Client VM (20.6-b01) for windows-x86 JRE (1.6.0_31-b05), built on Feb  3 2012 18:44:09 by "java_re" with MS VC++ 7.1 (VS2003)

time: Tue Feb 23 09:47:21 2016
elapsed time: 14 seconds
```

gchauvet, 2016-02-23:
Note: same issue in a non threaded execution way

jason-s, 2016-02-23:
The crash-JVM-upon-USB-disconnect is has been an issue with RXTX for years (I ran into it in 2009 and 2010) and has never been properly addressed to my knowledge. It's the reason why I stopped using RXTX completely and migrated to [PureJavaComm](http://www.sparetimelabs.com/purejavacomm/purejavacomm.php) which does not have this issue.

- http://marc.info/?l=rxtx&m=128896796230399&w=2

- http://stackoverflow.com/questions/2142918/java-serial-i-o-handling-usb-serial-connection-disconnection-in-a-robust-manner

I hope someone is able to fix this issue.

madhephaestus, 2016-02-23:
The work around i have used for a long time for this is:

* cache a copy of the data streams. don't just call get stream each time.
* when you detect the disconnect, set the stream to null and disconnect
* the read method will fail with a catchable exception when it tries to read from the null stream. 

I have a consumer serial device i ship with this lib, and the use case of pulling it before disconnecting was a primary concern. I have never dived into the JNI to attempt to pus the robustness into the place where the error occurs, but this workaround provides the expected results for my application.

jason-s, 2016-02-23:
>when you detect the disconnect, set the stream to null and disconnect

It's been several years since I was working with RXTX or NRJavaSerial, but from what I remember, this was possible in certain cases and not possible in others -- again, see http://marc.info/?l=rxtx&m=128896796230399&w=2. Low data rates (NOT baud rates -- you can have a 4.375Mbaud connection and only send a few hundred bytes per second if you want) made it easier to catch and recover; high data rates made it harder. There was a guy on the RXTX mailing list (Mariusz) that claimed he could always workaround the disconnect crash issue, but he was using low data rates and I was using high data rates. I was able to reproduce his method on low data rates, but unfortunately never got around to exploring in detail before I left my previous job along with my sample code.

gchauvet, 2016-02-24:
Using PureJavaComm solve my issue :+1: 
But I've small differences : timeout threshold work like Linux. I define it at 500ms as describe [here](https://msdn.microsoft.com/en-us/library/system.io.ports.serialport.readtimeout%28v=vs.110%29.aspx#Remarks)

madhephaestus, 2020-05-06:
this issue resolved as part of #75

closed 2020-05-06 by madhephaestus