NeuronRobotics/nrjavaserial #68: Ducplicate ressource in ditribution jar

gchauvet, 2016-02-23:
Some files (in meta-inf folder) are duplicated. When we want to use this library in a applet way, our JRE raise an exception because "LICENSE.txt" entry is duplicated in JAR file (this is not the single file).

GitoMat, 2016-05-11:
I have the same issue. "jarsigner" cannot process the jar file. The command fails:
`$ jarsigner -keystore store.jks -signedjar output/nrjavaserial-3.11.0.jar input/nrjavaserial-3.11.0.jar keyalias`
`jarsigner: unable to sign jar: java.util.zip.ZipException: duplicate entry: META-INF/LICENSE.txt`

The file is included twice:
`unzip -l nrjavaserial-3.11.0.jar | grep LICENSE.txt`
`11358  2014-01-05 18:01   META-INF/LICENSE.txt`
`11560  2013-06-07 23:49   META-INF/LICENSE.txt`

madhephaestus, 2016-05-13:
Let me know if that worked for you?

GitoMat, 2016-05-17:
Yes, signing works now, but as I discovered it already worked before this commit as LICENCE.txt wasn't in the Jar anymore.

closed 2020-04-09 by madhephaestus