NeuronRobotics/nrjavaserial #72: Arm64 v8hf support

xsnrg, 2016-03-10:
I didn't see an issue that called this out in particular, but is there any plan to support the new 64bit ARM v8 cpus?  The Odroid C2 sports this CPU and is in the wild (I have one).  If I understand correctly, the Pine64 that just started shipping is also of this arch.  OpenHAB uses nrjavaserial and as such cannot support (without some tinkering) serial devices on these new cpus.  They currently try to load the 32 bit library, and fail.

libNRJavaSerial.so: wrong ELF class: ELFCLASS32 (Possible cause: can't load IA 32-bit .so on a ARM-bit platform)

uname -a
Linux odroid64 3.14.29-29 #1 SMP PREEMPT Fri Feb 26 11:00:53 BRT 2016 aarch64 aarch64 aarch64 GNU/Linux

specs here:  http://www.hardkernel.com/main/products/prdt_info.php

Other mention of ARM64 here:  https://github.com/NeuronRobotics/nrjavaserial/issues/51

MrDOS, 2016-03-10:
Yep, I'd love to support them. The only reason we currently don't is lack of resources. I don't have any ARM64 devices on hand but I've been meaning to buy a couple of different ODROIDs to get some more ARM variants on hand for testing (I have access to ARMv5 and v6 but no v7 or v8). As far as the native code support is concerned, I don't suspect much effort to be involved: the code doesn't seem to explicitly call out ARM support anywhere, so it's hopefully just a matter of recompilation given that other 64-bit architectures are supported happily.

xsnrg, 2016-03-10:
> it's hopefully just a matter of recompilation given that other 64-bit architectures are supported happily.

I was hoping that was the case.  Thanks for the quick reply.  I'd be willing to test a build when the time comes.

bcraun, 2016-03-22:
Not sure if this is at all related, but I'm trying to recompile for ARM v7 and am successful with the compilation, but the Gradle "processResources" task is failing due to a file not found exception. The file is clearly present, and has the correct permissions as far as I can tell. Here's the stacktrace from the Gradle task that's failing. Any help is appreciated greatly. Thank you.

gradle build --stacktrace
:compileJava UP-TO-DATE
:processResources FAILED

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':processResources'.
> Could not copy file '/home/pi/nrjavaserial_ia32/nrjavaserial/src/main/resources/jvser.properties' to '/home/pi/nrjavaserial_ia32/nrjavaserial/build/resources/main/jvser.properties'.

* Try:
Run with --info or --debug option to get more log output.

* Exception is:
org.gradle.api.tasks.TaskExecutionException: Execution failed for task ':processResources'.
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.executeActions(ExecuteActionsTaskExecuter.java:72)
	at org.gradle.api.internal.tasks.execution.ExecuteActionsTaskExecuter.execute(ExecuteActionsTaskExecuter.java:49)
	at org.gradle.api.internal.tasks.execution.PostExecutionAnalysisTaskExecuter.execute(PostExecutionAnalysisTaskExecuter.java:34)
	at org.gradle.api.internal.changedetection.CacheLockHandlingTaskExecuter$1.run(CacheLockHandlingTaskExecuter.java:34)
	at org.gradle.internal.Factories$1.create(Factories.java:22)
	at org.gradle.cache.internal.DefaultCacheAccess.longRunningOperation(DefaultCacheAccess.java:179)
	at org.gradle.cache.internal.DefaultCacheAccess.longRunningOperation(DefaultCacheAccess.java:232)
	at org.gradle.cache.internal.DefaultPersistentDirectoryStore.longRunningOperation(DefaultPersistentDirectoryStore.java:142)
	at org.gradle.api.internal.changedetection.DefaultTaskArtifactStateCacheAccess.longRunningOperation(DefaultTaskArtifactStateCacheAccess.java:83)
	at org.gradle.api.internal.changedetection.CacheLockHandlingTaskExecuter.execute(CacheLockHandlingTaskExecuter.java:32)
	at org.gradle.api.internal.tasks.execution.SkipUpToDateTaskExecuter.execute(SkipUpToDateTaskExecuter.java:55)
	at org.gradle.api.internal.tasks.execution.ValidatingTaskExecuter.execute(ValidatingTaskExecuter.java:57)
	at org.gradle.api.internal.tasks.execution.SkipEmptySourceFilesTaskExecuter.execute(SkipEmptySourceFilesTaskExecuter.java:41)
	at org.gradle.api.internal.tasks.execution.SkipTaskWithNoActionsExecuter.execute(SkipTaskWithNoActionsExecuter.java:51)
	at org.gradle.api.internal.tasks.execution.SkipOnlyIfTaskExecuter.execute(SkipOnlyIfTaskExecuter.java:52)
	at org.gradle.api.internal.tasks.execution.ExecuteAtMostOnceTaskExecuter.execute(ExecuteAtMostOnceTaskExecuter.java:42)
	at org.gradle.api.internal.AbstractTask.executeWithoutThrowingTaskFailure(AbstractTask.java:275)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor.executeTask(DefaultTaskPlanExecutor.java:52)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor.processTask(DefaultTaskPlanExecutor.java:38)
	at org.gradle.execution.taskgraph.DefaultTaskPlanExecutor.process(DefaultTaskPlanExecutor.java:30)
	at org.gradle.execution.taskgraph.DefaultTaskGraphExecuter.execute(DefaultTaskGraphExecuter.java:84)
	at org.gradle.execution.SelectedTaskExecutionAction.execute(SelectedTaskExecutionAction.java:29)
	at org.gradle.execution.DefaultBuildExecuter.execute(DefaultBuildExecuter.java:61)
	at org.gradle.execution.DefaultBuildExecuter.access$200(DefaultBuildExecuter.java:23)
	at org.gradle.execution.DefaultBuildExecuter$2.proceed(DefaultBuildExecuter.java:67)
	at org.gradle.api.internal.changedetection.TaskCacheLockHandlingBuildExecuter$1.run(TaskCacheLockHandlingBuildExecuter.java:31)
	at org.gradle.internal.Factories$1.create(Factories.java:22)
	at org.gradle.cache.internal.DefaultCacheAccess.useCache(DefaultCacheAccess.java:124)
	at org.gradle.cache.internal.DefaultCacheAccess.useCache(DefaultCacheAccess.java:112)
	at org.gradle.cache.internal.DefaultPersistentDirectoryStore.useCache(DefaultPersistentDirectoryStore.java:134)
	at org.gradle.api.internal.changedetection.DefaultTaskArtifactStateCacheAccess.useCache(DefaultTaskArtifactStateCacheAccess.java:79)
	at org.gradle.api.internal.changedetection.TaskCacheLockHandlingBuildExecuter.execute(TaskCacheLockHandlingBuildExecuter.java:29)
	at org.gradle.execution.DefaultBuildExecuter.execute(DefaultBuildExecuter.java:61)
	at org.gradle.execution.DefaultBuildExecuter.access$200(DefaultBuildExecuter.java:23)
	at org.gradle.execution.DefaultBuildExecuter$2.proceed(DefaultBuildExecuter.java:67)
	at org.gradle.execution.DryRunBuildExecutionAction.execute(DryRunBuildExecutionAction.java:32)
	at org.gradle.execution.DefaultBuildExecuter.execute(DefaultBuildExecuter.java:61)
	at org.gradle.execution.DefaultBuildExecuter.execute(DefaultBuildExecuter.java:54)
	at org.gradle.initialization.DefaultGradleLauncher.doBuildStages(DefaultGradleLauncher.java:166)
	at org.gradle.initialization.DefaultGradleLauncher.doBuild(DefaultGradleLauncher.java:113)
	at org.gradle.initialization.DefaultGradleLauncher.run(DefaultGradleLauncher.java:81)
	at org.gradle.launcher.cli.ExecuteBuildAction.run(ExecuteBuildAction.java:38)
	at org.gradle.launcher.exec.InProcessGradleLauncherActionExecuter.execute(InProcessGradleLauncherActionExecuter.java:39)
	at org.gradle.launcher.exec.InProcessGradleLauncherActionExecuter.execute(InProcessGradleLauncherActionExecuter.java:25)
[... cut ...]
	at org.gradle.internal.nativeplatform.filesystem.LibCStat.getUnixMode(LibCStat.java:41)
	at org.gradle.internal.nativeplatform.filesystem.GenericFileSystem.getUnixMode(GenericFileSystem.java:62)
	at org.gradle.api.internal.file.DefaultFileTreeElement.getMode(DefaultFileTreeElement.java:66)
	at org.gradle.api.internal.file.AbstractFileTreeElement.copyTo(AbstractFileTreeElement.java:67)
	... 81 more

BUILD FAILED

Total time: 1 mins 12.806 secs
Makefile:29: recipe for target 'arm' failed
make: *** [arm] Error 1

kaikreuzer, 2016-03-23:
Is there any chance to have that included in the upcoming 3.12 release, potentially with https://github.com/NeuronRobotics/nrjavaserial/pull/71?

umiddelb, 2016-04-19:
This Makefile builds a native `libNRJavaSerial.so` on the aarch64 (PINE64 or ODROID-C2) (make sure that you have installed the `liblockdev1-dev` package first):
```
LINOBJ=build/fixup.o build/fuserImp.o build/SerialImp.o
LININCLUDE=-I/usr/lib/jvm/java-7-openjdk-arm64/include/ -I"./include" -I"./include/target"

CCLINARM=gcc $(LININCLUDE) -DLIBLOCKDEV -O3 -Wall -c -fmessage-length=0 -fPIC -MMD -MP
LINKLINARM=gcc -shared -fPIC

dirs:
        if (test -d build); then rm -rf build;mkdir build;else mkdir build; fi
        if (test -d resources/native);\
         then echo Native output directories seem to exist.;else\
         mkdir -p resources/native/linux/ARM/; \
         fi

arm:    dirs
        $(CCLINARM) src/fixup.c -o build/fixup.o
        $(CCLINARM) src/fuserImp.c -o build/fuserImp.o
        $(CCLINARM) src/SerialImp.c -o build/SerialImp.o
        $(LINKLINARM) -o"resources/native/linux/ARM/libNRJavaSerial.so" $(LINOBJ) -llockdev
        rm -rf build
```

xsnrg, 2016-04-20:
Thanks for the Makefile.  It appears that liblockdev1 is deprecated, replaced with flock:  https://launchpad.net/ubuntu/xenial/arm64/liblockdev1

The packages are still available when you hunt them down, but that will come to an end.  This is with Ubuntu xenial.

umiddelb, 2016-04-21:
You're welcome. I prefer using Debian Jessie on the C2 and PINE64 due to several issues in Ubuntu Xenial at this time (may be they are fixed until the official release date). Debian Jessie doesn't label liblockdev1 as deprecated.

Replacing liblockdev1 by flock might affect the original Makefile (this file may require some refactoring / modularization) as well.

closed 2020-04-09 by madhephaestus