NeuronRobotics/nrjavaserial #73: Detect fifo buffer underrun

TheNetStriker, 2016-03-21:
Is it possible to detect fifo buffer overflows with this library?

The reason why I'am asking is because of the changes in this issue: https://github.com/NeuronRobotics/nrjavaserial/issues/36

I suspect that my OpenHAB server is loosing some ZWave events because of this. Is there a way to check how often the serial port buffer did overflow?

madhephaestus, 2016-03-21:
The buffers are dynamic and should only "overflow" if your computer runs out of ram. I have pushed a steady 1Mbaud for days and days without any overflow on a ras pi. I would suggest using VisualVM to analyze your application's memory management because it sounds more like leaky memory than some buffer drop. If you are losing bytes in the serial stream, look at your physical layer and microcontroller code.

TheNetStriker, 2016-03-21:
There should be more than enough free ram on my BeagleBone Black, but it seams to make a difference if I use the new or the old version of nrjavaserial. (Some of the ZWave values ar not loading with the new version) I will also discuss this in the OpenHAB forum after some more testing.

TheNetStriker, 2016-03-21:
I forgot to ask, did you test this with the latest 3.12.0 version? (The issue was merged after 3.11.0 was released)

madhephaestus, 2020-05-06:
stale closing

closed 2020-05-06 by madhephaestus