NeuronRobotics/nrjavaserial #81: Not listing seril ports even though there's  a plenty of them

rafaelcn, 2016-06-03:
Hi, I was testing this lib on my computer (`linux amd 64`) to maybe replace the one that I daily use (RXTX) and I did a small example to list all the serial ports on my computer as:

```
package com.serialtest;

import gnu.io.NRSerialPort;

public class Main {

    public static void main(String[] args) {
        for(String s: NRSerialPort.getAvailableSerialPorts()) {
            System.out.println("Availible port(s): " + s);
        }
    }
}
```

But, though I have a lot of serial ports, as seen in:

```
[ranu@ranu-laptop ~]$ ls -l /dev/ttyS*
crw-rw---- 1 root dialout 4, 64 Jun  3 07:15 /dev/ttyS0
crw-rw---- 1 root dialout 4, 65 Jun  3 07:15 /dev/ttyS1
crw-rw---- 1 root dialout 4, 74 Jun  3 07:15 /dev/ttyS10
crw-rw---- 1 root dialout 4, 75 Jun  3 07:15 /dev/ttyS11
crw-rw---- 1 root dialout 4, 76 Jun  3 07:15 /dev/ttyS12
crw-rw---- 1 root dialout 4, 77 Jun  3 07:15 /dev/ttyS13
crw-rw---- 1 root dialout 4, 78 Jun  3 07:15 /dev/ttyS14
crw-rw---- 1 root dialout 4, 79 Jun  3 07:15 /dev/ttyS15
crw-rw---- 1 root dialout 4, 80 Jun  3 07:15 /dev/ttyS16
crw-rw---- 1 root dialout 4, 81 Jun  3 07:15 /dev/ttyS17
crw-rw---- 1 root dialout 4, 82 Jun  3 07:15 /dev/ttyS18
crw-rw---- 1 root dialout 4, 83 Jun  3 07:15 /dev/ttyS19
crw-rw---- 1 root dialout 4, 66 Jun  3 07:15 /dev/ttyS2
crw-rw---- 1 root dialout 4, 84 Jun  3 07:15 /dev/ttyS20
crw-rw---- 1 root dialout 4, 85 Jun  3 07:15 /dev/ttyS21
crw-rw---- 1 root dialout 4, 86 Jun  3 07:15 /dev/ttyS22
crw-rw---- 1 root dialout 4, 87 Jun  3 07:15 /dev/ttyS23
crw-rw---- 1 root dialout 4, 88 Jun  3 07:15 /dev/ttyS24
crw-rw---- 1 root dialout 4, 89 Jun  3 07:15 /dev/ttyS25
crw-rw---- 1 root dialout 4, 90 Jun  3 07:15 /dev/ttyS26
crw-rw---- 1 root dialout 4, 91 Jun  3 07:15 /dev/ttyS27
crw-rw---- 1 root dialout 4, 92 Jun  3 07:15 /dev/ttyS28
crw-rw---- 1 root dialout 4, 93 Jun  3 07:15 /dev/ttyS29
crw-rw---- 1 root dialout 4, 67 Jun  3 07:15 /dev/ttyS3
crw-rw---- 1 root dialout 4, 94 Jun  3 07:15 /dev/ttyS30
crw-rw---- 1 root dialout 4, 95 Jun  3 07:15 /dev/ttyS31
crw-rw---- 1 root dialout 4, 68 Jun  3 07:15 /dev/ttyS4
crw-rw---- 1 root dialout 4, 69 Jun  3 07:15 /dev/ttyS5
crw-rw---- 1 root dialout 4, 70 Jun  3 07:15 /dev/ttyS6
crw-rw---- 1 root dialout 4, 71 Jun  3 07:15 /dev/ttyS7
crw-rw---- 1 root dialout 4, 72 Jun  3 07:15 /dev/ttyS8
crw-rw---- 1 root dialout 4, 73 Jun  3 07:15 /dev/ttyS9
```

It lists no ports! I tested on a Windows machine and it does display a single serial port `COM1`, so I assumed that the problem may be in my computer, do you guys have any idea of what might be the problem?

madhephaestus, 2016-06-03:
this is a Linux file permissions issue. 

crw-rw---- 1 root dialout means its only accessible to users root and members of group dialout. add the user to the group dialout, or make a mySerial.rules file with something like:

SUBSYSTEM=="tty", ATTRS{product}=="*Bootloader*",			SYMLINK+="Bootloader%n",   SYMLINK+="Bootloader.%E{ID_SERIAL_SHORT}", 	MODE="0777", ENV{ID_MM_DEVICE_IGNORE}="1"

to automatically symlink, rename and add global access to specific serial ports.

rafaelcn, 2016-06-03:
I forgot to mention, I added myself to the dialout group as I suspected it was a permission issue:

```
[ranu@ranu-laptop ~]$ groups ranu
ranu : ranu adm dialout cdrom sudo dip plugdev lpadmin sambashare bumblebee
```

And I never heard of a `*.rules` file, thanks for the info, gonna look into it.

closed 2020-04-09 by madhephaestus