NeuronRobotics/nrjavaserial #83: Parametrize setSerialPortParams

savirsavir, 2016-06-22:
Too use this library you should add option to change serial port params, there is only baudrate and not other: data size, parity etc.

serial.setSerialPortParams(getBaud(), SerialPort.DATABITS_8, SerialPort.STOPBITS_1, SerialPort.PARITY_NONE);

andyrozman, 2016-06-22:
I had same problem a while ago, which is why I made change to library and added method getSerialPortInstance(), which returns RxtxPort instance, on which you can set everything. It would be better if we had all methods visible on main class, but at least now you can set it. 

We could either add all this methods to main class (NRSerialPort) or we could also add method setParameters, which would have as input parameter SerialSettings (or something), and would set everything that was set there... Or you can just use this solution with getSerialPortInstance() for now...

madhephaestus, 2020-05-06:
stale closing

closed 2020-05-06 by madhephaestus