NeuronRobotics/nrjavaserial #84: java.lang.UnsatisfiedLinkError on 3.12.0+

lunderhage, 2016-07-12:
Hi,

I get an UnsatisfiedLinkError when running my application on Raspberry Pi, starting from 3.12.0.

java.lang.UnsatisfiedLinkError: gnu.io.RXTXCommDriver.nativeGetVersion()Ljava/lang/String;
	at gnu.io.RXTXCommDriver.nativeGetVersion(Native Method)
	at gnu.io.RXTXCommDriver.<clinit>(RXTXCommDriver.java:105)
	at gnu.io.NRSerialPort.getAvailableSerialPorts(NRSerialPort.java:153)
	at se.lunderhage.pcr1000.backend.daemon.SerialPortUtils.getSerialPorts(SerialPortUtils.java:46)
	at ...

3.11.0 is the last version not having this problem, but there I can't use 38400 bps on the raspberry for some reason. (38400 works fine with minicom).

Will try to build the ARM binaries and see what happens...

Regards

Rallymen007, 2016-07-22:
Same issue here, I've tried building the ARM libraries and it didn't fix it. I can't test 3.11 however, as I need a karaf bundle.

Rallymen007, 2016-07-22:
False alert, I solved the issue by installing liblockdev1-dev. I don't know if it's possible to check for the presence of the lib and display a proper error message though.

madhephaestus, 2016-07-25:
The warning and conflict resolution would happen at the .deb packaging level. I just set myself up a new PPA, so ill look into bundling up this jar and making a system lib perhaps? It would require a lot more discussion in any case

closed 2016-07-25 by madhephaestus