NeuronRobotics/nrjavaserial #88: Nrjavaserial doesn't work in MAC

joseluisUEX, 2016-10-27:
I´ve used the library nrjavaserial-3.12.0.jar in my Mac (OS X 10.9.3) and the connection doesn't work. However, ports are detected using the following code:

       try{
			Enumeration<CommPortIdentifier> ports = CommPortIdentifier.getPortIdentifiers();
			while(ports.hasMoreElements())
				System.out.println(ports.nextElement().getName());
		}catch(Exception ex){
			ex.printStackTrace();
		}catch(Error er){
			er.printStackTrace();
		}

But when the program runs the following instruction

  serialPort = (SerialPort) portId.open(this.getClass().getName(),TIME_OUT);

The program returns an error.

If I change this library adding RXTXcomm-2.jar library instead, the program runs fine. So I suppose the problem is nrjavaserial library.

As we want the program runs in every system, we'd like to use nrjavaserial library. Does anybody know which is the problem?

MrDOS, 2016-11-02:
> The program returns an error.

Can you provide the error here?

joseluisUEX, 2016-11-03:
This is the message returned:

Exception in thread "main" java.lang.NoSuchMethodError: gnu.io.CommPortIdentifier.open(Ljava/lang/String;I)Lgnu/io/RXTXPort;
	at ConexionCOM.abrirConexion(ConexionCOM.java:94)
	at TestConexion.conectar(TestConexion.java:51)
	at TestConexion.main(TestConexion.java:21)
Experimental:  JNI_OnLoad called.

madhephaestus, 2020-04-09:
this is dependant on #163

madhephaestus, 2020-04-29:
check 5.0.0

closed 2020-04-29 by madhephaestus