NeuronRobotics/nrjavaserial #91: NRJavaSerial Not working on rPi2.

carryonrewardless, 2017-01-10:
My concern is similar to this one:
https://github.com/NeuronRobotics/nrjavaserial/issues/88
with a change from 3.9 to 3.12 in Openhab2 causing an error:
/dev/ttyAMA0 does not exist 
for the Razberry zwave card that sits on the Raspberry Pi 2 gpio pins.
Prior to the change the device was detected quite successfully using this Java option:
-Dgnu.io.rxtx.SerialPorts=/dev/ttyAMA0
Potential permissions issues have been thoroughly exhausted but I believe this issue may be specific to Debian Wheezy on the Raspberry Pi version 2. 
I am not a developer but hope I have provided enough information.

madhephaestus, 2017-02-07:
if you do ls -al /dev/ttyAMA0 && groups what do you get?

Does putty work? Any other serial communication tool?

jvalkeal, 2017-02-19:
@carryonrewardless I had similar issues with my RPi3 and when I poked around a little deeper actual error was.
```
java.lang.UnsatisfiedLinkError: /tmp/libNRJavaSerial_HF_pi_0/libNRJavaSerial_HF.so: liblockdev.so.1: cannot open shared object file: No such file or directory
```
Got it working by installing dependant library.
```
sudo apt-get install liblockdev1
```
Worth to check if this is a case in your system as well.

jvalkeal, 2017-02-19:
@madhephaestus is `liblockdev` really needed for arm or is it just a mistake that it's linked to library?

carryonrewardless, 2017-02-19:
Thanks for the comments. The liblockdev1 library does fix the problem on the RPi3 but not the RPi2 on Wheezy.  The answer came from cvanorman :

madhephaestus, 2017-02-20:
Soundls like this is all fixed now?

closed 2017-02-20 by madhephaestus