NeuronRobotics/nrjavaserial #92: Tags for release 3.12.1 unavailable

ndefeijter, 2017-01-18:
Tags for release 3.12.1 are not available in the git archive. We are using this version because it is the latest that can be found in Maven Central:
https://mvnrepository.com/artifact/com.neuronrobotics/nrjavaserial/3.12.1

Since we are having issues (endlessly blocking read), we want to inspect the code to:
- Inspect what might be causing the issue.
- Determine whether it is worthwhile downgrading to a previous version.

kaikreuzer, 2017-01-18:
See also https://github.com/NeuronRobotics/nrjavaserial/issues/44#issuecomment-246791847

Would be great to have proper tagging and release management done for the lib here on Github 👍

jvalkeal, 2017-02-18:
paging @madhephaestus while trying to figure out why I'm having trouble with arm7, indeed noticed that build is still in `3.12.0` even if `3.12.1` has been released. Would you mind to do updates as you're the one who changed build version last time?

madhephaestus, 2017-02-20:
Sorry for the lag on my part, fixed

closed 2017-02-20 by madhephaestus