NeuronRobotics/nrjavaserial #93: Java application crashes when launched as Executable

danielpeintner, 2017-03-09:
Hi,

I did run into a strange behavior. I use an application which reads some data from a serial port. No problem.. everything works fine. The "same" application launched as executable (EXE created with javapackager) crashed after trying to read data.

Operating system is Windows10, Java1.8.
nrjavaserial version 3.12.1 (same issue with older tried 3.10 version)

Looking at the crash dump shows the following.

```
Stack: [0x0000001109100000,0x0000001109200000],  sp=0x00000011091febd0,  free space=1018k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [libNRJavaSerial.dll+0x753d]
C  [libNRJavaSerial.dll+0x7876]
C  0x0000016a85006f1b

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  gnu.io.RXTXPort.readArray([BII)I+0
j  gnu.io.RXTXPort$SerialInputStream.read([BII)I+187
j  sun.nio.cs.StreamDecoder.readBytes()I+135
j  sun.nio.cs.StreamDecoder.implRead([CII)I+112
j  sun.nio.cs.StreamDecoder.read([CII)I+180
j  java.io.InputStreamReader.read([CII)I+7
j  java.io.BufferedReader.fill()V+145
J 35 C1 java.io.BufferedReader.readLine(Z)Ljava/lang/String; (304 bytes) @ 0x0000016a85106974 [0x0000016a851066a0+0x2d4]
j  java.io.BufferedReader.readLine()Ljava/lang/String;+2
j  org.peintner.timekeeper.TimeImportLayoutController$SimpleRead.run()V+276
j  java.lang.Thread.run()V+11
v  ~StubRoutines::call_stub
```

I can also provide the entire dump if it helps.

To me it is very strange that launching the JAR is fine while launching the EXE crashes.

Any idea/help appreciated!

Thanks for your work!

-- Daniel

madhephaestus, 2017-03-09:
I would look closer at what your javapackager configuration as it seems to be altering your working jar into a non working jar. I am not familiar with javapackager, but if it works as a jar and not after your packager touched it, i would much more likely blame the packager ;)

riedlse, 2018-04-13:
I have the same issue. Works fine when run in netbeans (and I assume with a java -jar) But when I use Advanced Installer to package, I get the crash. This just started happening so not sure it's advanced installer or java 161 in combination with nrjavaserial.
[hs_err_pid14560.log](https://github.com/NeuronRobotics/nrjavaserial/files/1908875/hs_err_pid14560.log)

closed 2020-04-09 by madhephaestus