NeuronRobotics/nrjavaserial #99: Internal library errors messages

RicardoFR, 2017-05-22:
Hi,

I have discovered that there are some errors that are displayed in console (for example when disconnecting the serial cable) but that I am not able to capture as an exception or to avoid that they are displayed in the console, any idea or suggestion?

Thank you !

crea-doo, 2017-07-18:
I'm also not happy with the way it works now. I use the library for hardware bindings running in an OSGi framework with a separate console and everytime an error occurs it clutters up the console.
I suggest to use some kind of logging framework to do proper logging and error handling. I've implemented this in a branch already but unfortunately not all messages are captured - some of them seem to come directly out of the C libraries. See https://github.com/crea-doo/nrjavaserial/tree/logging for details.

@madhephaestus: Do you have some hints, where these messages are sent out and how to catch these?

madhephaestus, 2017-07-18:
this was an issue that goes all the way back to RXTX. I would be ok with a new version of the binaries with all of the C level prints removed, but that comes with the headache of compiling all of the object files for a release. I had hoped that the CI builders would be ready at some point that @MrDOS was working on.

crea-doo, 2017-07-22:
So the goal would be to have an CI configuration first, which builds the libraries for all the supported platforms?! And then the other issues can be solved more easily.

Did you try the updated build scripts from https://github.com/NeuronRobotics/nrjavaserial/pull/71?

Maybe we should try to get some advanced Travis configuration (see https://docs.travis-ci.com/user/multi-os/)?
It looks like you can build on ubuntu (for cross compilation) and on osx images and then push the compiled libraries back into e.g. a separate branch (see e.g. http://blog.yuriy.tymch.uk/2013/03/sharing-travis-ci-generated-files.html just to get the idea)

madhephaestus, 2017-07-25:
Yeah, the multi os is exactly what we need. I think it stalled when the original script took too long to compile all three in the single instance. Maybe breaking them each out to the given OS would make this easiest.

crea-doo, 2017-07-27:
I tried the configuration in a separate branch (https://github.com/crea-doo/nrjavaserial/tree/travis-multi-os or https://travis-ci.org/crea-doo/nrjavaserial/builds/258276801) and have some success - see also my comments in https://github.com/NeuronRobotics/nrjavaserial/pull/71#issuecomment-318464341

madhephaestus, 2017-07-27:
Ok, i merged that PR for the updated build system, and have enabled TravisCI for this repo:

https://travis-ci.org/NeuronRobotics/nrjavaserial

madhephaestus, 2020-05-06:
I added a hardware disconnect event to #75

closed 2020-05-06 by madhephaestus