NeuronRobotics/nrjavaserial #105: NoClassDefFoundError

mvglasow, 2017-06-24:
After building according to the instructions, I get the following error when I try to run the example code:

```
Exception in thread "main" java.lang.NoClassDefFoundError: com/sun/jna/platform/win32/Win32Exception
	at gnu.io.NRSerialPort.getAvailableSerialPorts(NRSerialPort.java:153)
```

OS is Linux 64-bit, which explains why Win32 platform classes are not present, but makes it the more puzzling that these classes ever get referenced.

andyrozman, 2017-06-25:
You are (probably) missing dependencies. New version of nrjavaserial, requires you to have jna and jna-platform installed (4.2.2 version)

mvglasow, 2017-06-27:
I just checked (searched for `jna` and `libjna` in Synaptic) and have the following packages installed:

* libjna-jni (4.2.2-1)
* libjna-java (4.2.2-1)
* libjna-platform (4.2.2-1)

The one I'm missing is libjna-java-doc, but I wouldn’t expect that to do any harm. There’s also libjna-posix-java and libjna-posix-java-doc (both version 1.0.1-2), which are not installed, but the version number suggests they are not part of the JNA core.

PeteSL, 2017-07-30:
We ran into this and you need jna-platform.jar in your classpath for RXTXCommDriver to load.  As far as I can tell, that is the only place where JNA is used and only to get those Win32 definitions.  My question to the author is "Are you going to migrate all of the native code to JNA?" and "If not, why not?"  The jna-platform.jar is basically a large set of definitions (2 meg worth) while the base jna.jar has all of the operating code (yes, you can program using jna.jar without jna-platform.jar but not vice versa).  Having converted native Linux JNI code to JNA, I can say that while not trivial, it also was not difficult.  I have participated with the development of another Java serial interface that is 100% based on JNA.  The biggest issue is you need to use the Native library interfaces for performance reasons and that does carry some restrictions but that just means the developers need to understand the variances between native and non-native JNA methods and structures.  Personally, I would like to see a migration to JNA which would make your ongoing maintenance significantly easier (no more keeping up with various distros or architectures) as long as you use the Native library interface to be as close as possible to c performance.

closed 2020-05-06 by madhephaestus