NeuronRobotics/nrjavaserial #107: Null Pointer Exception with addEventListener on Spring Boot

ierturk, 2017-07-18:
I get the message along with `addEventListener`

> This is a bug, passed the ownership test above: NRSerialPort

java.lang.NullPointerException
	at gnu.io.NRSerialPort.addEventListener(NRSerialPort.java:212)
	at org.erturk.online.serial.milkmeter.CommMM.Init(CommMM.java:70)
	at org.ieee.ierturk.interfarm.CustomConfig.onApplicationEvent(CustomConfig.java:36)
	at org.ieee.ierturk.interfarm.CustomConfig.onApplicationEvent(CustomConfig.java:23)
	at org.springframework.context.event.SimpleApplicationEventMulticaster.invokeListener(SimpleApplicationEventMulticaster.java:167)
	at org.springframework.context.event.SimpleApplicationEventMulticaster.multicastEvent(SimpleApplicationEventMulticaster.java:139)
	at org.springframework.context.support.AbstractApplicationContext.publishEvent(AbstractApplicationContext.java:393)
	at org.springframework.context.support.AbstractApplicationContext.publishEvent(AbstractApplicationContext.java:347)
	at org.springframework.context.support.AbstractApplicationContext.finishRefresh(AbstractApplicationContext.java:883)
	at org.springframework.boot.context.embedded.EmbeddedWebApplicationContext.finishRefresh(EmbeddedWebApplicationContext.java:144)
	at org.springframework.context.support.AbstractApplicationContext.refresh(AbstractApplicationContext.java:546)
	at org.springframework.boot.context.embedded.EmbeddedWebApplicationContext.refresh(EmbeddedWebApplicationContext.java:122)
	at org.springframework.boot.SpringApplication.refresh(SpringApplication.java:693)
	at org.springframework.boot.SpringApplication.refreshContext(SpringApplication.java:360)
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:303)
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:1118)
	at org.springframework.boot.SpringApplication.run(SpringApplication.java:1107)
	at org.ieee.ierturk.interfarm.InterFarmApp.main(InterFarmApp.java:26)
This is a bug, passed the ownership test above: NRSerialPort

xxxxvelocityxxxx, 2018-02-06:
I'm also getting this when I try to add an event listener. My code is
```java    
 NRSerialPort serial;                 
 port = "COM5";
 baudRate = 38400;  
 serial = new NRSerialPort(port, baudRate);
 try {
    serial.addEventListener(new SerialPortEventListener() {
	@Override
	public void serialEvent(SerialPortEvent arg0) {
		System.out.println(arg0.getEventType());
	}
    });
} catch (TooManyListenersException e) {
	e.printStackTrace();
}
```
and it gives me
```java
Exception in thread "AWT-EventQueue-0" java.lang.NullPointerException
	at gnu.io.NRSerialPort.addEventListener(NRSerialPort.java:212)
	at station12.SerialCommunicate.<init>(SerialCommunicate.java:50)
	at station12.Station12Main.<init>(Station12Main.java:65)
	at station12.Station12Main.createAndShowGui(Station12Main.java:176)
	at station12.Station12Main.lambda$0(Station12Main.java:194)
	at java.awt.event.InvocationEvent.dispatch(Unknown Source)
	at java.awt.EventQueue.dispatchEventImpl(Unknown Source)
	at java.awt.EventQueue.access$500(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.awt.EventQueue$3.run(Unknown Source)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(Unknown Source)
	at java.awt.EventQueue.dispatchEvent(Unknown Source)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForFilter(Unknown Source)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.pumpEvents(Unknown Source)
	at java.awt.EventDispatchThread.run(Unknown Source)
```
on the addEventListener line

MrDOS, 2018-02-06:
There's no API documentation to make it clear, but you have to call `connect()` on the `NRSerialPort` instance before you can do just about anything else with it. ([`NRSerialPort.java:212`](https://github.com/NeuronRobotics/nrjavaserial/blob/9e7abd4cdf20a4173d5b17a16e153b978a34de13/src/main/java/gnu/io/NRSerialPort.java#L212) tries to use the `serial` instance variable which isn't set until after `connect()` has been called.)

Try this:

```
NRSerialPort serial;
port = "COM5";
baudRate = 38400;
serial = new NRSerialPort(port, baudRate);
serial.connect();
try {
    serial.addEventListener(new SerialPortEventListener() {
        @Override
        public void serialEvent(SerialPortEvent arg0) {
            System.out.println(arg0.getEventType());
        }
    });
} catch (TooManyListenersException e) {
    e.printStackTrace();
}
```

xxxxvelocityxxxx, 2018-02-06:
Thanks. That fixed it. Would you happen to know how I can detect if a com port becomes unavailable? I am checking for COM5, which is my arduino, and if I start my program with it disconnected, then connect it I am able to detect it, but when I disconnect the USB cable NRSerialPort.getAvailableSerialPorts() still tells me COM5 is available and I'm not getting any events or anything really to indicate it has been disconnected.

madhephaestus, 2020-05-06:
usage issue

closed 2020-05-06 by madhephaestus