NeuronRobotics/nrjavaserial #108: Lib crahes in OH2 on Synology libNRJavaSerial

matty67, 2017-08-20:
I´m using openhab 2.1.0
The included nrjavaserial 3.12.0 OH is not working.
So i tried some older nrjavaserial libs in runtime/lib folder.
The nrjavaserial versions 3.11.0, 3.10 and 3.9 are working.
The newer one with the missing native/linux/ARM_A8 dir are not working.
They are crahing the system. For example the exception for the included nrjavaserial 3.12.0 OH:

#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0x4b6d9c1c, pid=28575, tid=0x5215b470
#
# JRE version: Java(TM) SE Embedded Runtime Environment (8.0_121-b13) (build 1.8.0_121-b13)
# Java VM: Java HotSpot(TM) Embedded Client VM (25.121-b13 mixed mode linux-arm )
# Problematic frame:
# C  [libNRJavaSerial.so+0x9c1c]  JNI_OnLoad+0x28
#
# Core dump written. Default location: /volume1/public/oh2/userdata/core or core.28575
#
# An error report file with more information is saved as:
# /volume1/public/oh2/userdata/hs_err_pid28575.log
[thread 1270477936 also had an error]
[thread 1381086320 also had an error]
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
#

kaikreuzer, 2017-08-21:
> The included nrjavaserial 3.12.0 OH is not working.

Not sure if this is the right place to report as 3.12.0 OH is a fork and not an official project release.
Did you try with the official 3.12.1 or 3.13.0?

matty67, 2017-08-21:
Hi Kai. Thanks for your reply.
Yes you´re right. I think this isn´t the right place but I don´t know where to report it. Where I can do this?
I tried 3.12.1 and 3.13.0. Both not running. Error is the same like above.

kaikreuzer, 2017-08-21:
> I tried 3.12.1 and 3.13.0. Both not running. Error is the same like above. 

Well, in this case, this is the right place to report - but you didn't mention that anywhere above.

matty67, 2017-08-21:
Ok. I ´ve updated my comment.
"The newer one (also 3.12.1 and 3.13.0) ...."

b8rother, 2017-11-23:
@matty67
I'm experiencing the same problem. Could you please tell me how to include older versions (3.11.0 or 3.10.2) nrjavaserial lib to runtime/lib folder?
I have tried to put jar file in runtime/lib folder and runtime/lib/boot but it doesn't get loaded. I'm logged in via ssh karaf and looking at the bundles via bundle:list | grep nr but I can't see the package.
I have removed the nrjavaserial 3.12.0-OH bundle with bundle:uninstall nrjavaserial.

matty67, 2017-11-23:
I just put it into the runtime/lib folder. I used the enocean lib. This lib loads the nrjavaserial.
Don´t forget to put the serial device into the start.sh like this.
export EXTRA_JAVA_OPTS="-Dgnu.io.rxtx.SerialPorts=/dev/ttyUSB0:/dev/enocean"

b8rother, 2017-11-23:
Thank you for the reply. I'm not sure how to use your solution for my issue since I'm using Z-wave binding.
Which nrjavaserrial jar package are you using?
I have a clean install of openHAB-2.2.0.008-SNAPSHOT (I've also tried 2.1.0.008-DSM6) on Synology DS413j. I'm using Aeotec Z-wave z-stick Gen5 as USB/serial gateway for Z-wave sensors.

I have followed most of the tutorials which implements this device on synology but I had no luck to get it finally working. One of them which summarize all tasks can be found here: http://www.hagemann.ws/blog/add-java-8,-openhab-and-z-wave-to-a-synology-diskstation-os-6.1-virtual-machine.html

But since this is a nrjavaserial issue tracker I will not explain further but just say that whenever I try to add a Thing in OH2 JVM crashes. Here is the exception:

```
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGILL (0x4) at pc=0x4c5eac1c, pid=11609, tid=0x5166f470
#
# JRE version: Java(TM) SE Embedded Runtime Environment (8.0_151-b12) (build 1.8.0_151-b12)
# Java VM: Java HotSpot(TM) Embedded Client VM (25.151-b12 mixed mode linux-arm )
# Problematic frame:
# C  [libNRJavaSerial.so+0x9c1c]  JNI_OnLoad+0x28
#
# Core dump written. Default location: /volume1/@appstore/openHAB2/userdata/core or core.11609
#
# An error report file with more information is saved as:
# /volume1/@appstore/openHAB2/userdata/hs_err_pid11609.log
#
# If you would like to submit a bug report, please visit:
#   http://bugreport.java.com/bugreport/crash.jsp
# The crash happened outside the Java Virtual Machine in native code.
# See problematic frame for where to report the bug.
```
And here is the **hs_err_pid11609.log**

```
---------------  T H R E A D  ---------------

Current thread (0x52441800):  JavaThread "qtp1741886-205" [_thread_in_native, id=12201, stack(0x51620000,0x51670000)]

siginfo: si_signo: 4 (SIGILL), si_code: 1 (ILL_ILLOPC), si_addr: 0x4c5eac1c

Registers:
  r0  = 0x00000002
  r1  = 0x00000000
  r2  = 0x000001e4
  r3  = 0x4c5fd2a8
  r4  = 0x00000000
  r5  = 0x4eed50a0
  r6  = 0x52441934
  r7  = 0x5166cf64
  r8  = 0x5166cf68
  r9  = 0x52415f10
  r10 = 0x533b8390
  fp  = 0x4c5eabf4
  r12 = 0x4c5e52b0
  sp  = 0x5166cdc8
  lr  = 0x4c5eac18
  pc  = 0x4c5eac1c
  cpsr = 0x60000010

Top of Stack: (sp=0x5166cdc8)
0x5166cdc8:   00000000 408a6704 52441800 50c77060
0x5166cdd8:   4ac21840 4074bbf4 5166ced0 00000000
0x5166cde8:   5166cf50 5166ce38 4374f770 4b6aca90
0x5166cdf8:   40743cac 4ac21840 52441800 52d93754
0x5166ce08:   4ac00b28 52441800 5166ce1c 405b05e8
0x5166ce18:   5166ce2c 4047e690 52441800 4374edb8
0x5166ce28:   5166ceac 40453f04 4ac21840 52441800
0x5166ce38:   0000000e 00000000 00000001 00000013 

Instructions: (pc=0x4c5eac1c)
0x4c5eabfc:   e08f3003 e92d4010 e7933002 e5830000
0x4c5eac0c:   e59f0018 e08f0000 ebffe32c e3a00002
0x4c5eac1c:   e3400001 e8bd8010 000123fc 000001e4
[... 112 lines cut ...]
j  org.eclipse.jetty.server.session.SessionHandler.doHandle(Ljava/lang/String;Lorg/eclipse/jetty/server/Request;Ljavax/servlet/http/HttpServletRequest;Ljavax/servlet/http/HttpServletResponse;)V+68
j  org.eclipse.jetty.server.handler.ContextHandler.doHandle(Ljava/lang/String;Lorg/eclipse/jetty/server/Request;Ljavax/servlet/http/HttpServletRequest;Ljavax/servlet/http/HttpServletResponse;)V+349
j  org.ops4j.pax.web.service.jetty.internal.HttpServiceContext.doHandle(Ljava/lang/String;Lorg/eclipse/jetty/server/Request;Ljavax/servlet/http/HttpServletRequest;Ljavax/servlet/http/HttpServletResponse;)V+50
j  org.eclipse.jetty.servlet.ServletHandler.doScope(Ljava/lang/String;Lorg/eclipse/jetty/server/Request;Ljavax/servlet/http/HttpServletRequest;Ljavax/servlet/http/HttpServletResponse;)V+274
j  org.eclipse.jetty.server.session.SessionHandler.doScope(Ljava/lang/String;Lorg/eclipse/jetty/server/Request;Ljavax/servlet/http/HttpServletRequest;Ljavax/servlet/http/HttpServletResponse;)V+257
j  org.eclipse.jetty.server.handler.ContextHandler.doScope(Ljava/lang/String;Lorg/eclipse/jetty/server/Request;Ljavax/servlet/http/HttpServletRequest;Ljavax/servlet/http/HttpServletResponse;)V+453
j  org.eclipse.jetty.server.handler.ScopedHandler.handle(Ljava/lang/String;Lorg/eclipse/jetty/server/Request;Ljavax/servlet/http/HttpServletRequest;Ljavax/servlet/http/HttpServletResponse;)V+20
J 4732 C1 org.ops4j.pax.web.service.jetty.internal.JettyServerHandlerCollection.handle(Ljava/lang/String;Lorg/eclipse/jetty/server/Request;Ljavax/servlet/http/HttpServletRequest;Ljavax/servlet/http/HttpServletResponse;)V (218 bytes) @ 0x40dd64b0 [0x40dd6160+0x350]
j  org.eclipse.jetty.server.handler.HandlerWrapper.handle(Ljava/lang/String;Lorg/eclipse/jetty/server/Request;Ljavax/servlet/http/HttpServletRequest;Ljavax/servlet/http/HttpServletResponse;)V+18
j  org.eclipse.jetty.server.Server.handle(Lorg/eclipse/jetty/server/HttpChannel;)V+141
j  org.eclipse.jetty.server.HttpChannel.handle()Z+313
j  org.eclipse.jetty.server.HttpConnection.onFillable()V+127
j  org.eclipse.jetty.io.AbstractConnection$ReadCallback.succeeded()V+4
j  org.eclipse.jetty.io.FillInterest.fillable()V+61
j  org.eclipse.jetty.io.SelectChannelEndPoint$2.run()V+7
j  org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.executeProduceConsume()V+313
j  org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.produceConsume()V+28
j  org.eclipse.jetty.util.thread.strategy.ExecuteProduceConsume.run()V+106
j  org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(Ljava/lang/Runnable;)V+1
j  org.eclipse.jetty.util.thread.QueuedThreadPool$2.run()V+104
j  java.lang.Thread.run()V+11
v  ~StubRoutines::call_stub
```

Since I haven't seen similar problem except yours I'm starting to think that there could be something wrong with the hardware itself (Z-Wave Usb Stick).

matty67, 2017-11-23:
In your stacktrace I see the same problem that I had. The stub could not be found.
`j  gnu.io.RXTXCommDriver.<clinit>()V+39`
`v  ~StubRoutines::call_stub`

I´ve used the nrjavaserial-3.11.0.jar.
Don´t install the serial binding in oh2. Put the nrjavaserial-3.11.0.jar into the runtime/lib folder.
If you wan to use a newser version, have a look in the jar file if the folder /native/linux/ARM_A8 exists.
If the folder not exists, you´ll get  the above error.

b8rother, 2017-11-23:
I've copied the nrjavaserial-3.11.0.jar (http://repo1.maven.org/maven2/com/neuronrobotics/nrjavaserial/3.11.0/nrjavaserial-3.11.0.jar) to the runtime/lib. I haven't installed the serial binding in OH2 but I guess that the Z-Wave binding installs the nrjavaserial-3.12.0.OH as a dependency. So I uninstalled it with bundle:uninstall 25 (current Id of nrjavaseria-3.12.0l).
Then I stopped and started again OH (sh start_debug.sh) and in log:tail I got:

```
20:25:21.537 [ERROR] [org.eclipse.smarthome.config.serial  ] - FrameworkEvent ERROR - org.eclipse.smarthome.config.serial
org.osgi.framework.BundleException: Could not resolve module: org.eclipse.smarthome.config.serial [107]
  Unresolved requirement: Import-Package: gnu.io

        at org.eclipse.osgi.container.Module.start(Module.java:444) [?:?]
        at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1620) [?:?]
        at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.incStartLevel(ModuleContainer.java:1600) [?:?]
        at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.doContainerStartLevel(ModuleContainer.java:1571) [?:?]
        at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1514) [?:?]
        at org.eclipse.osgi.container.ModuleContainer$ContainerStartLevel.dispatchEvent(ModuleContainer.java:1) [?:?]
        at org.eclipse.osgi.framework.eventmgr.EventManager.dispatchEvent(EventManager.java:230) [?:?]
        at org.eclipse.osgi.framework.eventmgr.EventManager$EventThread.run(EventManager.java:340) [?:?]
```
 ZWave binding is declared (in karaf console) only as **Installed** and not **Active** because it's missing **gnu.io** package from nrjavaserial. Am I correctly unistalling version 3.12.0? Can you guess why nrjavaserial.3.11.0 is not being loaded from runtime/lib? I have set chown to openhab2:users and chmod 077 to the 3.11.0 jar.

I'm sorry to bother you so much but you are currently the only one willing and capable to solve this issue. Thank you again.

matty67, 2017-11-23:
Ok. Look under /userdata/etc/ in the file config.properties.
The gnu.io have to be added to the file, like I show you in the next lines.
#
# Extra packages appended after standard packages
#
org.osgi.framework.system.packages.extra = \
    org.apache.karaf.branding, \
    org.apache.karaf.jaas.boot.principal, \
    org.apache.karaf.jaas.boot, \
    sun.misc, \
    gnu.io, \

b8rother, 2017-11-23:
That does it. Everything is working, Zwave controller is recognized and working. 
I hope that it will help somebody else with the same issue. 
I don't know how to express my gratitude other than to say thank you once again and that I owe a beer or two.

matty67, 2017-11-23:
That sounds great. That its working and the beer that is waiting for me. haha.

madhephaestus, 2020-04-09:
let me know if the official 3.19.0 fixes this?

closed 2020-04-09 by madhephaestus