NeuronRobotics/nrjavaserial #112: JVM crash on macOS.

alex-vas, 2017-10-22:
[hs_err_pid59556.log](https://github.com/NeuronRobotics/nrjavaserial/files/1404617/hs_err_pid59556.log)

alex-vas, 2017-10-22:
Seen it once. Other time works seemingly ok.

pilhuhn, 2018-01-03:
I have seen a different crash

```
get_java_var: invalid file descriptor
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x000000011420cb08, pid=45717, tid=0x0000000000005903
#
# JRE version: Java(TM) SE Runtime Environment (8.0_151-b12) (build 1.8.0_151-b12)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.151-b12 mixed mode bsd-amd64 compressed oops)
# Problematic frame:
# C  [libNRJavaSerial.jnilib+0x5b08]  read_byte_array+0x58
#

```

kaikreuzer, 2018-09-13:
Just came across the very same:
```
get_java_var: invalid file descriptor
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x0000000110690b08, pid=31464, tid=0x0000000000010d07
#
# JRE version: Java(TM) SE Runtime Environment (8.0_112-b16) (build 1.8.0_112-b16)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.112-b16 mixed mode bsd-amd64 compressed oops)
# Problematic frame:
# C  [libNRJavaSerial.jnilib+0x5b08]  read_byte_array+0x58
#
```

madhephaestus, 2020-04-09:
Depends on #163

madhephaestus, 2020-04-29:
check new binaries in 5.0.0

commit af6139da (2024-02-09): Fix segfault if file descriptor unavailable

The `get_java_var_long` function returns 0 in several failure modes,
e.g. if a file descriptor is unavailable. [1]

However, one of the call sites is missing the result check, which causes
a JVM segfault if the return value is 0. The segfault occurs on
dereferencing the pointer: [2]

```c
eis->eventflags[SPE_DATA_AVAILABLE]
```

Add a result value check, throwing a proper IOException if it is 0.

See also similar issue #59. [3]

Fixes #112 [4], #136 [5] and #242 [6].

[1]: https://github.com/NeuronRobotics/nrjavaserial/blob/0df8b60485a56d7698b711832 [... cut]
closed 2020-04-29 by madhephaestus