NeuronRobotics/nrjavaserial #116: Crash with Java 9.0.4

BerndInBS, 2018-02-20:
Using the library with Java 9.0.4 causes a crash when reading data from the serial port. This also happens when I use the example from the README file. Switching to Java 8 fixes this crash.

Current thread (0x00000226ec844000):  JavaThread "JavaFX Application Thread" [_thread_in_native, id=23688, stack(0x000000fa31400000,0x000000fa31500000)]

Stack: [0x000000fa31400000,0x000000fa31500000],  sp=0x000000fa314faa60,  free space=1002k
Native frames: (J=compiled Java code, j=interpreted, Vv=VM code, C=native code)
C  [libNRJavaSerial.dll+0x753d]
C  [libNRJavaSerial.dll+0x7876]
C  0x00000226c535921c

Java frames: (J=compiled Java code, j=interpreted, Vv=VM code)
j  gnu.io.RXTXPort.readArray([BII)I+0
j  gnu.io.RXTXPort$SerialInputStream.read([BII)I+187
j  gnu.io.RXTXPort$SerialInputStream.read([B)I+38

mizoguch-ken, 2018-03-02:
It is not the best method, but please try the following

1. nrjavaserial/src/main/c/include/SerialImp.h
```
 ...
 long get_java_var_long (JNIEnv *, jobject, char *, char *);
 ...
```
to
```
...
long long get_java_var_long (JNIEnv *, jobject, char *, char *);
...
```

2. nrjavaserial/src/main/c/src/SerialImp.c
```
...
long get_java_var_long( JNIEnv *env, jobject jobj, char *id, char *type )
{
	long result = 0;
...
	if ( !strcmp( type, "J" ) ) {
	  result = (long)( (*env)->GetLongField( env, jobj, jfd ) );
	} else {
	  result = (size_t) ( (*env)->GetIntField( env, jobj, jfd ) );
	}
...
```
to
```
...
long long get_java_var_long( JNIEnv *env, jobject jobj, char *id, char *type )
{
	long long result = 0;
...
	if ( !strcmp( type, "J" ) ) {
	  result = (long long)( (*env)->GetLongField( env, jobj, jfd ) );
	} else {
	  result = (size_t) ( (*env)->GetIntField( env, jobj, jfd ) );
	}
...
```

3. rebuild native code
4. rebuild jar

riedlse, 2018-04-18:
I did the fix above and it works fine now. Should it also be using jint for the JNI int callback?
I also forked and added some creates that allowed you to set additional parameters on instantiation and would like to fold those back in from my fork.

Maia-Everett, 2018-07-25:
This fix works for me as well and fixes the crash under 64-bit Windows Java 10. @mizoguch-ken, why do you think it's "not the best method"? I assume that the issue is data truncation, because `long` is 32-bit under 64-bit Windows, and it's converted to pointers elsewhere in the code. In this case, using `jlong` to represent a Java `long`, which is always 64-bit, does seem the correct solution to me.

mizoguch-ken, 2018-07-29:
Sorry for the late reply.
There are two reasons why I think "not the best method".
First, if it is decided to process a pointer, creating and using a pointer manipulation function is easy to understand and use at low cost.
Second, it may be different from the author 's guess as it does not understand the source' s intent.

dougmeredith, 2019-01-22:
I believe I'm encountering the same problem. It seems to occur on Windows only with any version of Java newer than 8.

I've done some testing and this occurs on 64-bit Windows 10 Using AdoptOpenJDK versions 9, 10, and 11. It also occurs using Oracle's Java 10

There seems to be no problem using Java 8 on Windows. I've also had no problem with AdoptOpenJDK 11 on 64-bin Linux.

FHDumay, 2019-05-25:
I am having the same issue on windows 7 with Oracle JDK/JRE 8 and 11. The simple example works, the issue i am having is that i am trying to read the DataInputStream in a seprarate thread

I attached a dump for both java 8 and 11 , perhaps it may help.
[hs_err_pid5316.log](https://github.com/NeuronRobotics/nrjavaserial/files/3219390/hs_err_pid5316.log)
[hs_err_pid11488.log](https://github.com/NeuronRobotics/nrjavaserial/files/3219389/hs_err_pid11488.log)

mizoguch-ken, 2019-07-09:
It is something which changed to jlong ​​in dll of windows

[nrjavaserial-3.15.0.zip](https://github.com/NeuronRobotics/nrjavaserial/files/3368496/nrjavaserial-3.15.0.zip)

FHDumay, 2019-07-11:
Thanks for the update. I will try the jar provided in the zip this weekend
to see if it works now.

zehoss, 2020-02-18:
Hi,
are you gonna fix this issue in maven repository?

madhephaestus, 2020-04-09:
this issue is fixed and published in 3.19.0

closed 2020-04-09 by madhephaestus