NeuronRobotics/nrjavaserial #118: "make windows" command difficulty

tstarker, 2018-03-29:
Hello, Sorry for the clutter/mess/english.

I was trying to run "make windows" on windows 7, but it has been a bit hard to make it work:

I'm most sure there's must be an easier way to do this, but I'm not used to gradle or make, just java programming.

I've had to install gradle and followed the readme on installing the tdm-gcc, here's what I needed to do to get a successful build:

`G:\nfc\commlib\nrjavaserial [master =]> make windows
make -C .\src\main\c windowsLocal
make[1]: Entering directory 'G:/nfc/commlib/nrjavaserial/src/main/c'
"C:\TDM-GCC-64\bin\gcc.exe" -I".\include" -I".\include\target" -I".\include\wind
ows" -I"C:\TDM-GCC-64\include" -I"D:/java-32/jdk1.7.0_25/include" -I"D:/java-32/
jdk1.7.0_25/include/win32" -m64 -O3 -Wall -DBUILD_DLL -c src/windows/init.c -o b
uild/init.o
src/windows/init.c:6:17: fatal error: jni.h: No such file or directory
compilation terminated.
Makefile:106: recipe for target 'windowsLocal' failed
make[1]: *** [windowsLocal] Error 1
make[1]: Leaving directory 'G:/nfc/commlib/nrjavaserial/src/main/c'
Makefile:5: recipe for target 'windows' failed
make: *** [windows] Error 2`

I've saw a strange directory there, "D:/java-32/jdk1.7.0_25/include" , I've decided to download latest jdk 1.7 and put it in the same location it's asking for, "D:/java-32/jdk1.7.0_25/", so I ran into a different error this time:

`G:\nfc\commlib\nrjavaserial [master =]> make windows
make -C .\src\main\c windowsLocal
make[1]: Entering directory 'G:/nfc/commlib/nrjavaserial/src/main/c'
"C:\TDM-GCC-64\bin\gcc.exe" -I".\include" -I".\include\target" -I".\include\wind
ows" -I"C:\TDM-GCC-64\include" -I"D:/java-32/jdk1.7.0_25/include" -I"D:/java-32/
jdk1.7.0_25/include/win32" -m64 -O3 -Wall -DBUILD_DLL -c src/windows/init.c -o b
uild/init.o
Assembler messages:
Fatal error: can't create build/init.o: No such file or directory
Makefile:106: recipe for target 'windowsLocal' failed
make[1]: *** [windowsLocal] Error 1
make[1]: Leaving directory 'G:/nfc/commlib/nrjavaserial/src/main/c'
Makefile:5: recipe for target 'windows' failed
make: *** [windows] Error 2`

So, I've created the "build" folder: G:/nfc/commlib/nrjavaserial/src/main/c/build

Then, ran the command again, this time got a bit of compilation warnings, and at the end:
(full log: https://pastebin.com/V3fqs1FT)
`[...]
"C:\TDM-GCC-64\bin\gcc.exe" -shared -Wl,--add-stdcall-alias -DBUILD_DLL -o"resou
rces/native/windows/x86_64/libNRJavaSerial.dll" build/SerialImp.o build/fuserImp
.o build/fixup.o build/termios.o build/init.o
"C:\TDM-GCC-32\bin\gcc.exe" -I".\include" -I".\include\target" -I".\include\wind
ows" -I"C:\TDM-GCC-32\include" -I"D:/java-32/jdk1.7.0_25/include" -I"D:/java-32/
jdk1.7.0_25/include/win32" -m32 -O3 -Wall -DBUILD_DLL -c src/windows/init.c -o b
uild/init.o
/usr/bin/sh: C:\TDM-GCC-32\bin\gcc.exe: No such file or directory
Makefile:106: recipe for target 'windowsLocal' failed
make[1]: *** [windowsLocal] Error 127
make[1]: Leaving directory 'G:/nfc/commlib/nrjavaserial/src/main/c'
Makefile:5: recipe for target 'windows' failed
make: *** [windows] Error 2
`

So, I've decided to install tdm-gcc-32, and it finally worked, "BUILD SUCCESSFUL".

My question is, am I missing something basic there? Some basic concept perhaps? I had to create directories and install gradle, tdm-gcc-64 and tdm-gcc-32, and also instal jdk 1.7 on the directory it was being shown, but is there a simpler way of compiling it next time without needing to do all of these workarounds? (creating directory, installing on specific folder)

PS: I'm mostly sure that "make windows" should be run on windows only, because I tried to run it in ubuntu and got 
`tstarker@localserver:~/nfc/nrjavaserial$ make windows
make -C .\src\main\c windowsLocal
make[1]: *** .srcmainc: No such file or directory.  Stop.
Makefile:5: recipe for target 'windows' failed
make: *** [windows] Error 2
`

PS2: I also tried to run "make wine" on ubuntu 16.04 and got:
`tstarker@localserver:~/nfc/nrjavaserial$ make wine
make -C src/main/c windows
make[1]: Entering directory '/home/tstarker/nfc/nrjavaserial/src/main/c'
if (test -d build); then rm -rf build;mkdir build;else mkdir build; fi
if (test -d resources/native);\
 then echo Native output directories seem to exist.;else\
 mkdir -p resources/native/linux/x86_32/;\
 mkdir -p resources/native/linux/x86_64/;\
 mkdir -p resources/native/linux/ARM/; \
 mkdir -p resources/native/linux/ARM_A8/; \
 mkdir -p resources/native/linux/ARM_ANDROID/; \
 mkdir -p resources/native/linux/PPC/; \
 mkdir -p resources/native/windows/x86_32/; \
 mkdir -p resources/native/windows/x86_64/; \
 mkdir -p resources/native/osx/; \
 fi
Native output directories seem to exist.
if (test -d resources/native/freebsd);\
 then echo "FreeBSD native dir exists"; else\
 mkdir -p resources/native/freebsd/x86_64/; \
 mkdir -p resources/native/freebsd/x86_32/; \
fi
FreeBSD native dir exists
export WINEPREFIX=/home/hephaestus/.wine/
wine "C:\MinGW\bin\gcc.exe" -I".\include" -I".\include\target" -I".\include\windows" -I"C:\MinGW\include" -I"D:/java-32/jdk1.7.0_25/include" -I"D:/java-32/jdk1.7.0_25/include\win32" -m32 -O3 -Wall -DBUILD_DLL src/windows/init.c -o build/init.o
wine: cannot find 'C:\MinGW\bin\gcc.exe'
Makefile:123: recipe for target 'windows' failed
make[1]: *** [windows] Error 2
make[1]: Leaving directory '/home/tstarker/nfc/nrjavaserial/src/main/c'
Makefile:8: recipe for target 'wine' failed
make: *** [wine] Error 2
`

andyrozman, 2018-03-29:
Why are you trying to build the project? Did you make any changes to code? 

If not, use maven/gradle and download the artifact. There were just few changes made after last version 3.13.0, was created. 

I never built the binaries myself, I did port project to use java5 in the past, and even then I didn't recompile the binaries, I just compiled java code and added all binaries supplied into jar file.

tstarker, 2018-03-29:
Ah, I see, thanks. As I said, I never used gradle/maven in the past, never knew that you could download the project already built, so I thought I'd need to compile it to use. I was simply following the readme instructions.

closed 2018-04-01 by tstarker