NeuronRobotics/nrjavaserial #120: java.lang.NoSuchFieldError: SIZE thrown while loading gnu.io.RXTXCommDriver

Wuifal, 2018-04-29:
Hey guys,

I'm using nrjavaserial with netbeans and everything works fine, as long I start my code from netbeans. I've tested two ways for the natives directory 1: I've added it to the classes directory and 2: just added it as source. Both works like a charm.

But when I load the jar with java.exe, I receive the exception above. Advapi32Util.registryGetValues is throwing it.

PeteSL, 2018-11-21:
This is caused by an incompatibility between the jna.jar and the jna-platform.jar you are using.  The JNA folks removed Pointer.SIZE in JNA 5.0 (current is 5.1) and replaced it with Native.POINTER_SIZE for commonality of classes in JNA.  That said, since NRJavaSerial uses jna-platform for its Windows support, be sure your jna.jar and jna-properties.jar are the same version.

madhephaestus, 2020-04-09:
OS/Java configuration, no change in library, closing

closed 2020-04-09 by madhephaestus