NeuronRobotics/nrjavaserial #123: Error building the jar in Windows

afernandezmlt, 2018-07-26:
I had an error trying to build the jar in windows, googling a bit I found out that on line 121 and 125 of build.gradle there was an error with the variables ossrhUsername and ossrhPassword.
When I change the line 
`authentication(userName: ossrhUsername, password: ossrhPassword)`
to 
`authentication(userName: hasProperty('ossrhUsername')?ossrhUsername:'', password: hasProperty('ossrhPassword')?ossrhPassword:'')`

the build works well.
Maybe there was reported previously but I didn't find it... 
Hope this helps.

tobidelbruck, 2018-11-12:
Thanks, that fixed it for me too!

mdelouwere, 2020-04-09:
Great fix,

worked for me ass well.

madhephaestus, 2020-04-09:
oopsie, this issue should never came up. That section is uesd for publishing to sonnotype. It makes reference to fields that exist in my secret credentials file, which is never comitted to git. I sym-link it into the build directory to publish. build.gradle should never have been pushed with that section uncommented.

closed 2020-04-09 by madhephaestus