NeuronRobotics/nrjavaserial #127: NRSerialPort.getAvailableSerialPorts() is empty in 3.14, wasn't in 3.13

pseudo555, 2018-09-24:
Hello,

NRSerialPort.getAvailableSerialPorts() is supposed to return all serial port available.

The following code: 
        for(String s: NRSerialPort.getAvailableSerialPorts()) {
            System.out.println("Availible port(s): " + s);
        }
Displays nothing using version 3.14.

In 3.13 (or previous, i was on 3.11 before), the same code (no other change expect nrjavaserial version) produces: 
Availible port(s): COM5
Availible port(s): COM3
Availible port(s): COM1 

So i don't know what broken that in those commits:
https://github.com/NeuronRobotics/nrjavaserial/compare/3.13.0...master

I'm using java 8.0 u60 x32 on win10 1803 x64 computer.

kaikreuzer, 2018-09-24:
This is a known issue on Windows - https://github.com/NeuronRobotics/nrjavaserial/pull/121 is supposed to fix it.
Unfortunately, it seems that this project is hardly maintained, so that such critical fixes in a release are not addressed :-/

pseudo555, 2018-09-24:
Thank's for your really quick answer.

From what i read, this should work as a workaround:
RXTXCommDriver RXTXDriver = new RXTXCommDriver();
RXTXDriver.initialize();
Set<String> x = RXTXDriver.getPortIdentifiers();

But in my case, X is still empty. what did i do wrong ?

kaikreuzer, 2018-09-25:
> the original code i used also works once jna & jna-platform are set as dependencies.

Right, this is the workaround I stated in https://github.com/NeuronRobotics/nrjavaserial/pull/121#issuecomment-410652896.

madhephaestus, 2018-11-16:
Is this fixed in the new 3.15.0?

madhephaestus, 2020-04-09:
this all should be fixed as of 3.19.0

closed 2020-04-09 by madhephaestus