NeuronRobotics/nrjavaserial #130: Could not get unknown property 'ossrhUsername' for object of type org.gradle.api.publication.maven.internal.deployer.DefaultGroovyMavenDeployer

WiMetrixDev, 2018-09-26:
I am building RXTX 64 bits for Ubuntu 18.04 LTS.

I have installed Gradle 4.10.2, JDK 10. On running `gradle build` I get this.

`Starting a Gradle Daemon (subsequent builds will be faster)`
`FAILURE: Build failed with an exception.`
`Where:`
`Build file '/home/mahmadmujtaba/Temp/nrjavaserial/build.gradle' line: 121`
`What went wrong:`
`A problem occurred evaluating root project 'nrjavaserial'.`
`> Could not get unknown property 'ossrhUsername' for object of type` `org.gradle.api.publication.maven.internal.deployer.DefaultGroovyMavenDeployer.`
`* Try:`
`Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.`
`* Get more help at https://help.gradle.org`
`Deprecated Gradle features were used in this build, making it incompatible with Gradle 5.0.`
`Use '--warning-mode all' to show the individual deprecation warnings.`
`See https://docs.gradle.org/4.10.2/userguide/command_line_interface.html#sec:command_line_warnings`

`BUILD FAILED in 6s`

What could be the issue here?

tobidelbruck, 2018-11-12:
See https://github.com/NeuronRobotics/nrjavaserial/issues/123 for the fix

madhephaestus, 2018-11-16:
this is fixed in master

closed 2018-11-16 by madhephaestus