NeuronRobotics/nrjavaserial #136: Crashes caused by libNRJavaSerial

splatch, 2018-12-03:
We have run into crashes while using nrjavaserial, especially when we try to restart our app which is using serial port:

```
get_java_var: invalid file descriptor
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x00007f25f62e47d5, pid=4123, tid=0x00007f25f60da700
#
# JRE version: Java(TM) SE Runtime Environment (8.0_192-b12) (build 1.8.0_192-b12)
# Java VM: Java HotSpot(TM) 64-Bit Server VM (25.192-b12 mixed mode linux-amd64 compressed oops)
# Problematic frame:
# C  [libNRJavaSerial.so+0x77d5]  read_byte_array+0x55
#
# Core dump written. Default location: openhab-2.3.0/userdata/core or core.4123
#
# An error report file with more information is saved as:
# openhab-2.3.0/userdata/hs_err_pid4123.log
Compiled method (c1)   78940 8272  s!   3       gnu.io.RXTXPort$SerialInputStream::read (98 bytes)
 total in heap  [0x00007f279e7cd250,0x00007f279e7cefe8] = 7576
 relocation     [0x00007f279e7cd378,0x00007f279e7cd4f8] = 384
 main code      [0x00007f279e7cd500,0x00007f279e7ce420] = 3872
 stub code      [0x00007f279e7ce420,0x00007f279e7ce558] = 312
 oops           [0x00007f279e7ce558,0x00007f279e7ce560] = 8
 metadata       [0x00007f279e7ce560,0x00007f279e7ce5e0] = 128
 scopes data    [0x00007f279e7ce5e0,0x00007f279e7cebe8] = 1544
 scopes pcs     [0x00007f279e7cebe8,0x00007f279e7ceec8] = 736
 dependencies   [0x00007f279e7ceec8,0x00007f279e7ceef0] = 40
 handler table  [0x00007f279e7ceef0,0x00007f279e7cef60] = 112
 nul chk table  [0x00007f279e7cef60,0x00007f279e7cefe8] = 136
```
Another crash which happened on other machine:
```
get_java_var: invalid file descriptor
#
# A fatal error has been detected by the Java Runtime Environment:
#
#  SIGSEGV (0xb) at pc=0x60e9b302, pid=4885, tid=0x5b2af470
#
# JRE version: OpenJDK Runtime Environment (8.0_152-b76) (build 1.8.0_152-b76)
# Java VM: OpenJDK Client VM (25.152-b76 mixed mode, Evaluation linux-aarch32 )
# Problematic frame:
# C  [libNRJavaSerial_HF.so+0x5302]  read_byte_array+0x31
#
# Failed to write core dump. Core dumps have been disabled. To enable core dumping, try "ulimit -c unlimited" before starting Java again
#
# An error report file with more information is saved as:
# /var/lib/openhab2/hs_err_pid4885.log
#
```

loesler, 2019-02-03:
Hello,

I can confirm this behaviour. The runtime environment crashes during the connection on Win10/Java 11. I got an error, i.e. `GetCommMOdemStatus failed!`. 

/Micha

wborn, 2019-03-25:
This issue is about a JVM crash with Java 8 on Linux (amd64/aarch32).

The library crashing the JVM with Java 11 on Windows is another known issue, see  https://github.com/NeuronRobotics/nrjavaserial/issues/131.

madhephaestus, 2020-04-09:
both of these should be addressed in 3.19.0

closed 2020-04-09 by madhephaestus