NeuronRobotics/nrjavaserial #139: No support for illumos, but have patch?

sjorge, 2019-02-09:
While trying to use OpenHAB on illumos (SmartOS distro, more specific) I noticed there is no support in nrjavaserial so I cannot connect to my zwave serial device.

I made a quick and dirty hack to get the native bindings and have a patch:

```
--- a/src/main/c/Makefile
+++ b/src/main/c/Makefile
@@ -51,6 +51,16 @@ LINKBSD32=c++ -m32 -shared
 CCBSD64=cc $(BSDINCLUDE) -O3 -Wall -c -fmessage-length=0 -fPIC -m64 -MMD
 LINKBSD64=c++ -m64 -shared

+ILMOBJ=build/fixup.o build/fuserImp.o build/SerialImp.o
+ILMINCLUDE=-I"./include" -I"./include/target" -I/opt/local/java/openjdk8/include/ -I/opt/local/java/openjdk8/include/solaris/
+
+CCILM32=cc $(ILMINCLUDE) -O3 -Wall -c -fmessage-length=0 -fPIC -m32 -MMD
+LINKILM32=c++ -m32 -shared
+
+
+CCILM64=cc $(ILMINCLUDE) -O3 -Wall -c -fmessage-length=0 -fPIC -m64 -MMD
+LINKILM64=c++ -m64 -shared
+

 #Wine Windows
 WINLINKOPT = -shared -Wl,--add-stdcall-alias -DBUILD_DLL
@@ -103,6 +113,11 @@ dirs:
         mkdir -p resources/native/freebsd/x86_64/; \
         mkdir -p resources/native/freebsd/x86_32/; \
        fi
+       if (test -d resources/native/illumos);\
+        then echo "illumos native dir exists"; else\
+        mkdir -p resources/native/illumos/x86_64/; \
+        mkdir -p resources/native/illumos/x86_32/; \
+       fi

 windowsLocal:
        $(CCWIN64N) -c src/windows/init.c -o build/init.o
@@ -171,6 +186,22 @@ freebsd64:dirs
        #rm "resources/native/freebsd/x86_64/libNRJavaSerial.so"
        $(LINKBSD64) -o"resources/native/freebsd/x86_64/libNRJavaSerial.so" $(LINOBJ)
        rm build/*
+illumos:illumos32 illumos64
+       echo all illumos ok!
+illumos32:dirs
+       $(CCILM32) src/fixup.c -o build/fixup.o
+       $(CCILM32) src/fuserImp.c -o build/fuserImp.o
+       $(CCILM32) src/SerialImp.c -o build/SerialImp.o
+       #rm "resources/native/illumos/x86_32/libNRJavaSerial.so"
+       $(LINKILM32) -o"resources/native/illumos/x86_32/libNRJavaSerial.so" $(LINOBJ)
+       rm build/*
+illumos64:dirs
+       $(CCILM64) src/fixup.c -o build/fixup.o
+       $(CCILM64) src/fuserImp.c -o build/fuserImp.o
+       $(CCILM64) src/SerialImp.c -o build/SerialImp.o
+       #rm "resources/native/illumos/x86_64/libNRJavaSerial.so"
+       $(LINKILM64) -o"resources/native/illumos/x86_64/libNRJavaSerial.so" $(LINOBJ)
+       rm build/*

 .PHONY: arm
 arm:
```

Once I know for sure this is working, would there be interesting in a PR?

sjorge, 2019-02-09:
illumos is fork of opensolaris, so I think I might need to rename illumos to solaris everywhere, pending testing with OpenHAB

sjorge, 2019-02-09:
Looks like some more patching is needed, I'll look into it.

sjorge, 2019-02-10:
I've update the diff in the first comment
```
2019-02-10 11:19:01.434 [INFO ] [ing.zwave.handler.ZWaveSerialHandler] - Connecting to serial port '/dev/cua/0'
2019-02-10 11:19:01.455 [INFO ] [ing.zwave.handler.ZWaveSerialHandler] - Serial port is initialized
2019-02-10 11:19:01.480 [INFO ] [ve.internal.protocol.ZWaveController] - Starting ZWave controller
2019-02-10 11:19:01.481 [INFO ] [ve.internal.protocol.ZWaveController] - ZWave timeout is set to 5000ms. Soft reset is false
```

Openhab now reports zwave as working! After compiling wit these patches and dropping the new library in /opt/openhab/addons!

Shall I open a PR?

sjorge, 2019-02-13:
Just confirmed, openhab is seeing my zwave sensor perfectly with this patch applied.

madhephaestus, 2020-04-09:
cool, submit a PR and add toolchain-build instructions to the readme for building under ubuntu 18.04.