NeuronRobotics/nrjavaserial #143: SerialInputStream does not throw IOException

t5keeper, 2019-05-10:
gnu.io.RXTXPort$SerialInputStream.read(byte[]) does not throws IOException when called on closed port.

t5keeper, 2019-05-10:
gnu.io.RXTXPort$SerialInputStream.write(byte[]) has the same problem.

Gerry33, 2019-07-15:
Same here: 
Reading on a removed device does throw lots of errors "Error 0x5 at src/windows/termios.c(2725): Access denied", but no Exception to catch

madhephaestus, 2020-05-06:
Use the https://github.com/NeuronRobotics/nrjavaserial/releases/tag/5.0.3-prerelease nrjavaserial-5.0.3-pre.jar and this code example:

https://github.com/NeuronRobotics/nrjavaserial/blob/issue75-serialEventExitsOnHardwareLoss/test/src/test/ReadTest.java#L39

to detect when the hardware is disconnected.

closed 2020-05-06 by madhephaestus