NeuronRobotics/nrjavaserial #149: how to convert byte to original format

jrajauriya, 2019-08-29:
we are using jSerialComm library. first we getting the garbage data after that i'm able to convert hex code. but not able to convert original string. thanks in advance...

			```
         byte[] readBuffer = new byte[comPort.bytesAvailable()];
	  int numRead = comPort.readBytes(readBuffer, readBuffer.length);
	   String s1 = new String(readBuffer);
	    char c1[] = s1.toCharArray();					                    
            for(char ch : c1)
             {
                System.out.print((int)ch + ", ");//convert to hex code
	    }

 output:
```
3,3,3,66,10,34,6,d,3,3,6,34,6,3e,4,6,6,d,3f,3,3,40,3,3f,3,40,3,3,

we want to expected output:
<SOH>BD<STX>SNO<HT>1<LF>DATE<HT>256<LF>4<HT>17<HT><HT>27<HT><HT>37<HT>

mizoguch-ken, 2019-09-11:
How about changing from
```
System.out.print((int) b + ",");
```
to
```
System.out.print((char) b + ",");
```

mizoguch-ken, 2019-09-13:
I think that byte data and expected value do not match.
Is the communication setting correct?

jrajauriya, 2019-09-14:
yes, we set the parameters `comPort.setComPortParameters(9600, 8, 1, SerialPort.NO_PARITY);` according to machine manual.

mizoguch-ken, 2019-09-14:
Is it possible to get the actual settings instead of the manual?

jrajauriya, 2019-09-16:
@mizoguch-ken yes it is possible, we got same output from com port to without set baud rate, data bits and parity.

mizoguch-ken, 2019-09-18:
From the expected results, I think it is probably a LIS Interface.
If so, Can I view Configure LIS settings with DM Configuration Tool?

madhephaestus, 2020-04-09:
not an issue in the library, closing

closed 2020-04-09 by madhephaestus