NeuronRobotics/nrjavaserial #153: HUPCL is unconditionally cleared in configure_port

a3f, 2019-12-09:
The Linux default is that new ports have the `HUPCL` flag (hang up) set, so RTS is released when the last user of the port closes the device file. rxtx always reconfigures new ports with
```
ttyset.c_cflag = CREAD | CS8 | CLOCAL;
```
which overrides the `HUPCL` flag. Here is an example how this breaks HW that expects RTS to be inactive when the serial port is unused:
https://forum.arduino.cc/index.php?topic=46144.0

I've a similar setup where a modem expects RTS to be deasserted when the host doesn't communicate with it. This needs to work even if the application terminates
abnormally, so I use the same fix locally, but would like to have the functionality upstream as well.

As I guess there might be *nix systems where `HUPCL` is not a default, I could draft a patch that doesn't remove `HUPCL` if it's already set for a port. This would still break setups, where the user runs rxtx, sets RTS to active, close the application and expects the RTS to stay asserted. I don't know how common such a use case is though.

Thoughts?

---
Note: I am not using your fork, but from a cursory look that part of the code is identical. So I figured I should ask for feedback on the approach here as well, as it seems to be the most active fork.

madhephaestus, 2019-12-09:
can you make it configurable so the default behavior is the same as the legacy, but adding this functionality as a user setting?

a3f, 2019-12-09:
That would work too. Do you have an example of an existing user setting that could be taken as a template?

a3f, 2020-02-21:
I'll close this issue for now. Should we implement the user setting on top of this fork, I'll open a pull request.

Cheers,

closed 2020-02-21 by a3f