NeuronRobotics/nrjavaserial #155: more-configurat

hauntingEcho, 2020-04-04:


madhephaestus, 2020-04-04:
To specify the java.io.tmpdir System property, you can invoke the JVM as follows:
```
java -Djava.io.tmpdir=/path/to/tmpdir
```
at runtime for your application with strange mount permissions.

closed 2020-04-04 by madhephaestus

hauntingEcho, 2020-04-09:
to clarify a bit, the use case for this is a web interface control for a sensor network.  As web-facing things tend to be a target for attack, setting `noexec` on Java's general `java.io.tmpdir` makes vulnerabilities more difficult to exploit.  I'm aware that the value of `java.io.tmpdir` can be changed - the question in this ticket was about disconnecting the piece which needs to be able to execute generated code (in `NativeResource`) from the general Java temp directory (which should not really be executing code coming from a web interface).  Apologies if my original post was unclear.