NeuronRobotics/nrjavaserial #158: Gradle 6.2 incompatible with osgi plugin

fwolter, 2020-04-08:
Thanks for merging #157 and updating the liblockdev dependency!

With Gradle 6.0 the osgi plugin became obsolete. See https://github.com/bndtools/bnd/blob/master/biz.aQute.bnd.gradle/README.md#replacing-use-of-gradles-deprecated-osgi-plugin

`FAILURE: Build failed with an exception.

* Where:
Build file '/home/fabian/nrjavaserial/build.gradle' line: 2

* What went wrong:
A problem occurred evaluating root project 'nrjavaserial'.
> Plugin with id 'osgi' not found.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1s
`

wborn, 2020-04-08:
I also tried the new 4.0.1 version with openHAB but it doesn't seem to have a usable MANIFEST.MF  @madhephaestus, causing exceptions in Karaf:

```
org.apache.karaf.features.internal.util.MultiException: Error:
	Unable to create resource for bundle mvn:com.neuronrobotics/nrjavaserial/4.0.1
	at org.apache.karaf.features.internal.download.impl.MavenDownloadManager$MavenDownloader.<init>(MavenDownloadManager.java:91)
	at org.apache.karaf.features.internal.download.impl.MavenDownloadManager.createDownloader(MavenDownloadManager.java:72)
	at org.apache.karaf.features.internal.region.Subsystem.downloadBundles(Subsystem.java:457)
	at org.apache.karaf.features.internal.region.Subsystem.downloadBundles(Subsystem.java:452)
	at org.apache.karaf.features.internal.region.SubsystemResolver.resolve(SubsystemResolver.java:224)
	at org.apache.karaf.features.internal.service.Deployer.deploy(Deployer.java:393)
	at org.apache.karaf.features.internal.service.FeaturesServiceImpl.doProvision(FeaturesServiceImpl.java:1062)
	at org.apache.karaf.features.internal.service.FeaturesServiceImpl.lambda$doProvisionInThread$13(FeaturesServiceImpl.java:998)
	at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
	at java.base/java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1128)
	at java.base/java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:628)
	at java.base/java.lang.Thread.run(Thread.java:834)
	Suppressed: java.lang.Exception: Unable to create resource for bundle mvn:com.neuronrobotics/nrjavaserial/4.0.1
		at org.apache.karaf.features.internal.region.Subsystem.createResource(Subsystem.java:824)
		at org.apache.karaf.features.internal.region.Subsystem.lambda$downloadBundles$1(Subsystem.java:499)
		at org.apache.karaf.features.internal.download.impl.MavenDownloadManager$MavenDownloader.lambda$download$0(MavenDownloadManager.java:138)
		at org.apache.karaf.features.internal.download.impl.DefaultFuture.notifyListener(DefaultFuture.java:350)
		at org.apache.karaf.features.internal.download.impl.DefaultFuture.notifyListeners(DefaultFuture.java:335)
		at org.apache.karaf.features.internal.download.impl.DefaultFuture.setValue(DefaultFuture.java:259)
		at org.apache.karaf.features.internal.download.impl.AbstractDownloadTask.setFile(AbstractDownloadTask.java:61)
		at org.apache.karaf.features.internal.download.impl.AbstractRetryableDownloadTask.run(AbstractRetryableDownloadTask.java:61)
		at java.base/java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:515)
		at java.base/java.util.concurrent.FutureTask.run(FutureTask.java:264)
		at java.base/java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:304)
		... 3 more
	Caused by: org.osgi.framework.BundleException: Unable to build resource for mvn:com.neuronrobotics/nrjavaserial/4.0.1: Unsupported 'Bundle-ManifestVersion' value: 1
		at org.apache.felix.utils.resource.ResourceBuilder.build(ResourceBuilder.java:82)
		at org.apache.felix.utils.resource.ResourceBuilder.build(ResourceBuilder.java:71)
		at org.apache.karaf.features.internal.region.Subsystem.createResource(Subsystem.java:822)
		... 13 more
	Caused by: org.osgi.framework.BundleException: Unsupported 'Bundle-ManifestVersion' value: 1
		at org.apache.felix.utils.resource.ResourceBuilder.doBuild(ResourceBuilder.java:90)
		at org.apache.felix.utils.resource.ResourceBuilder.build(ResourceBuilder.java:80)
		... 15 more
```

If you look at the manifest it's:

```
Manifest-Version: 1.0
Created-By: Neuron Robotics Cooperative
Specification-Title: nrjavaserial
Specification-Version: 4.0.1
Specification-Vendor: Neuron Robotics Cooperative
Implementation-Title: nrjavaserial
Implementation-Version: 4.0.1
Implementation-Vendor: Neuron Robotics Cooperative
```

whereas with 3.15.0 it previously was:

```
Manifest-Version: 1.0
Implementation-Title: nrjavaserial
Bundle-SymbolicName: com.neuronrobotics.nrjavaserial
Implementation-Version: 3.15.0
Bundle-ManifestVersion: 2
Bnd-LastModified: 1530282443000
Specification-Vendor: Neuron Robotics Cooperative
Specification-Title: nrjavaserial
Import-Package: com.sun.jna.platform.win32;resolution:=optional,org.ap
 ache.commons.net.telnet;resolution:=optional;version="[3.3,4)"
Require-Capability: osgi.ee;filter:="(&(osgi.ee=JavaSE)(version=1.6))"
Tool: Bnd-3.2.0.201605172007
Implementation-Vendor: Neuron Robotics Cooperative
Export-Package: gnu.io;version="3.15.0",gnu.io.rfc2217;version="3.15.0
 ";uses:="gnu.io,org.apache.commons.net.telnet"
Bundle-Name: nrjavaserial
Bundle-Version: 3.15.0
Created-By: Neuron Robotics Cooperative
Specification-Version: 3.15.0
```

madhephaestus, 2020-04-08:
the Import-Package section was also throwing errors in gradle 6.2. Java 11+ wasnt added to gradle until 6.x afaik, so it looks like java 11 and the osgi support are incompatible at the moment. I do not know anything about osgi, just merged in from devs that used it.

madhephaestus, 2020-04-08:
I tried to switch to the new version:

```
jar {
	jar.duplicatesStrategy = DuplicatesStrategy.EXCLUDE
	manifest {
		attributes(
			"Created-By": "Commonwealth Robotics Cooperative",
			"Specification-Title": props."app.name",
			"Specification-Version": props."app.version",
			"Specification-Vendor": "Commonwealth Robotics Cooperative",
			"Implementation-Title": props."app.name",
			"Implementation-Version" : props."app.version",
			"Implementation-Vendor": "Commonwealth Robotics Cooperative",
		)
	}
	bnd ( 'Import-Package': 'com.sun.jna.platform.win32;resolution:=optional,org.apache.commons.net.telnet;resolution:=optional,!gnu.io*,*')
	
}
```

But i still get this error:

```
* Where:
Build file '/home/hephaestus/git/nrjavaserial/build.gradle' line: 60

* What went wrong:
A problem occurred evaluating root project 'nrjavaserial'.
> Could not find method bnd() for arguments [{Import-Package=com.sun.jna.platform.win32;resolution:=optional,org.apache.commons.net.telnet;resolution:=optional,!gnu.io*,*}] on task ':jar' of type org.gradle.api.tasks.bundling.Jar.

```

madhephaestus, 2020-04-08:
Fixed in 4.20. and 3.18.0

closed 2020-04-08 by madhephaestus