NeuronRobotics/nrjavaserial #160: build sources does not work for rpm bound linux

Reissner, 2020-04-08:
I have suse linux 64. 
If i understand right, it is not just sufficient to build the jar, 
the sources must be built also. 
As suse is rpm based, had problem to include liblockfile. 
After that i had problems finding jni.h. 
A look at the `Makefile` shows that this is not updated to java 15. 

On the other hand, you provided dependency on a maven plugin. 
I just stolen the jar from `https://repo.maven.apache.org/maven2/com/neuronrobotics/nrjavaserial/4.0.1/`. 
And it worked. 
I imagine this is only because you also used x86_64 to build it. 
So it is not very reliable what i do.... 

I think it would be helpful to provide a project for the maven plugin.

madhephaestus, 2020-04-08:
The plan is to have the binaries built in CI. @MrDOS was working on that a while ago. I am going through and fixing up the build, so we can re-start that process if he was interested :)

madhephaestus, 2020-04-08:
In direct answer for you for local builds, either use a CI server or spool up an ubuntu VM to build.

closed 2020-04-08 by madhephaestus

Reissner, 2020-04-08:
good idea