NeuronRobotics/nrjavaserial #164: FreeBSD Binaries out of date

madhephaestus, 2020-04-09:
Someone needs to compile the 3.19.0 sources for FreeBSD

RafalLukawiecki, 2020-05-04:
FreeBSD here. How can I help get them recompiled?

wborn, 2020-05-05:
See [building native code](https://github.com/NeuronRobotics/nrjavaserial/#building-native-code) in the README. You can find the FreeBSD targets in the [Makefile](https://github.com/NeuronRobotics/nrjavaserial/blob/6d78ad7146ec0b97587c0c8820db8fd36a380a82/Makefile#L20-L28). Make sure to run it on the master branch so the libraries contain all the latest changes. Then create a PR that updates the recompiled native library/libraries in [native/freebsd](https://github.com/NeuronRobotics/nrjavaserial/tree/master/src/main/c/resources/native/freebsd).

RafalLukawiecki, 2020-05-05:
Thanks for the explanation @wborn, you made it clear how to share the binaries back with you. The only thing I am unsure about is how compiling the master branch will yield the 3.20.0 version. From what I can see, and having tried building the jar using gradle, it yielded version 5.0.0. I have not tried compiling the native code yet, so perhaps this is a moot point and 3.20 will just pop out somehow.

wborn, 2020-05-05:
When @madhephaestus created this issue 3.20.0 was the latest version.
The 4.x versions were compiled with Java 11 and thus required Java 11+ only.
The 5.x are compiled with Java 8 and work with Java 8-11+

We'll also switch to version 5.x before openHAB 3 is released.

RafalLukawiecki, 2020-05-05:
Between I and another OH user who might get there before me we will get you FreeBSD binaries of 5.0.0. I will separately report the issue with lockfiles in openhab-core, and I cross my fingers and hope that I can somehow recompile the lot to solve the problem I have in 2.5.4. I will link to this when the issue is ready.

madhephaestus, 2020-05-05:
the native sources shouldn't have changes since 3.20.0, but compile against 5.0.0 is also good. 

More importantly, i would like to get build instructions for ubuntu 18.04+ to cross compile these binaries. We are trying to set up CI to build all the platforms of native binaries, so the cross-compile on Linux is a goal.

wborn, 2020-05-05:
Maybe @markvk can also help with this?
The current FreeBSD libraries were added in https://github.com/NeuronRobotics/nrjavaserial/pull/62.

RafalLukawiecki, 2020-05-05:
I just recompiled master against the two FreeBSD targets (32 and 64 bit), see the pull request #170. Let me know if you need any other artefacts made on FreeBSD. I hope this somehow helps in the quest to solve some of the OH 2.5 nrjavaserial issues, like non-working RFC2217 ports (https://github.com/openhab/openhab-core/issues/1462) and breaking Java runtime due to lockfiles (https://github.com/openhab/openhab-core/issues/1461).

RafalLukawiecki, 2020-05-05:
May I ask a beginner's question, please? If I wanted to test if the newer version/binaries of nrjavaserial make any difference to the two issues that I am trying to solve, can I just swap out some files from within the OpenHAB runtime, or do I need to recompile the whole lot?

madhephaestus, 2020-05-05:
@RafalLukawiecki thank you, ill push out an update soon!

As for testing in OpenHAB, i think that question is more for them.

madhephaestus, 2020-05-05:
this should be resolved as of 5.0.2 in Maven

RafalLukawiecki, 2020-05-06:
@wborn, now that 5.0.2 with the new native code is on Maven, when will it make to https://openhab.jfrog.io/openhab/libs-release/com/neuronrobotics/nrjavaserial/ —is that transfer automatic? Thanks for guiding me on this.

madhephaestus, 2020-05-06:
It is synced to Maven Central and is availible now.

RafalLukawiecki, 2020-05-06:
Thank you, @madhephaestus! Much appreciated. On another note, I (and to some extent @wborn) are wondering if the issues I am seeing under FreeBSD might be related to the way nrjavaserial deals with locking. I have done some preliminary testing using yesterday evening build which I have described [here](https://github.com/openhab/openhab-core/issues/1461#issuecomment-624552032) and [here](https://github.com/openhab/openhab-core/issues/1462#issuecomment-624549923). Perhaps I should report them as issues against this repo?

madhephaestus, 2020-05-09:
I am reopening this because i added a C level feature that should be added to FreeBSD as well. THe feature is explained here: https://github.com/NeuronRobotics/nrjavaserial/releases/tag/5.1.0

RafalLukawiecki, 2020-05-09:
Would you like me to recompile now?

madhephaestus, 2020-05-09:
Yes please :)

closed 2020-05-09 by madhephaestus

RafalLukawiecki, 2020-05-09:
Pleasure! I'd like to ask for a spot of help in debugging nrjavaserial under FreeBSD in a few moments, if you don't mind... I can create an issue for that.

madhephaestus, 2020-05-09:
FreeBSD binaries up to date as of 5.1.1

PR #170 (2020-05-05): Recompiled native libraries for FreeBSD, on FreeBSD 11.3-RELEASE-p8
Which should resolve #164.