NeuronRobotics/nrjavaserial #165: PPC binaries are out of date

madhephaestus, 2020-04-09:
Someone needs to compile the PPC binaries against the 3.19.0 sources

madhephaestus, 2020-04-09:
Instructions for how to install the correct toolchain for this would be helpful. The ususla path is to have the makefile ensure the toolchain is installed, and ssuming that you are building on an ubuntu system means you can add the toolchain with apt. The windows binaries run in wine, and the other platforms are cross-compiles on a standard x86-64 system.

MrDOS, 2020-06-08:
Can you recall why the Makefile compiles for PowerPC SPE ([e500-series](https://en.wikipedia.org/wiki/PowerPC_e500)), not regular PowerPC? Support for powerpcspe has been [removed from GCC 9](https://gcc.gnu.org/git/?p=gcc.git;a=commitdiff;h=b31d0348ddada49453e3edaaf93a423fdc61dc79). And there doesn't appear to be any degree of binary compatibility between PowerPC and PowerPC SPE. I suspect the current compiler invocation is in error. However, because it's passed the scream test for literally years (i.e., no one has ever complained about it), I suspect we could just drop it altogether and no one would notice or mind.

closed 2021-01-20 by MrDOS