NeuronRobotics/nrjavaserial #167: Version 3.20.0 is Java 11+ only

wborn, 2020-04-15:
Today I also tried nrjavaserial 3.20.0 on another project still on Java 8, but it seems to be compiled for Java 11 @madhephaestus.

The Maven build errors with:

```
[ERROR]   bad class file: /home/wouter/.m2/repository/com/neuronrobotics/nrjavaserial/3.20.0/nrjavaserial-3.20.0.jar(gnu/io/CommPortIdentifier.class)
[ERROR]     class file has wrong version 55.0, should be 52.0
[ERROR]     Please remove or make sure it appears in the correct subdirectory of the classpath.
```

Wikipedia has a [table that explains these numbers](https://en.wikipedia.org/wiki/Java_class_file#General_layout).

Version 3.19.0 does seem to work with Java 8.

madhephaestus, 2020-04-15:
should be fixed in 3.20.1

closed 2020-04-15 by madhephaestus

wborn, 2020-04-15:
Thanks again! It's fixed. :+1: