NeuronRobotics/nrjavaserial #169: nrjavaserial needs a logo

madhephaestus, 2020-05-05:
We need a project logo. It would be nice to have a logo to use to reference this project.

celsou, 2021-06-11:
Hello.

I was looking at the issues in this repository and I came across this issue by chance. I decided to create some quick drawings in Inkscape as a logo suggestion.
If you like one of these two logos, you can use it freely under Creative Commons CC0 (public domain license).
I hope you like this images.

P.S.: I'm using Google Translate.

**Logo 1**
Font: DejaVu Sans

PNG version:
[image]
SVG source (inside the ZIP):
[NR logo 1.svg.zip](https://github.com/NeuronRobotics/nrjavaserial/files/6635189/NR.logo.1.svg.zip)

**Logo 2**
Font: Liberation Mono
PNG version:
[image]
SVG source (inside the ZIP):
[NR logo 2.svg.zip](https://github.com/NeuronRobotics/nrjavaserial/files/6635197/NR.logo.2.svg.zip)

celsou, 2021-06-12:
I will redesign these logos with a USB port instead of DB9. This also gave me an idea for another logo suggestion. When I finish I will post it here.

madhephaestus, 2021-06-12:
You are so awesome!!

celsou, 2021-06-15:
I redid the logos with a USB connector instead of DB9. Here are the images: 

PNG version:
[image] [image]

SVG version (inside ZIPs):
[NR logo 1 USB.zip](https://github.com/NeuronRobotics/nrjavaserial/files/6658701/NR.logo.1.USB.zip)
[NR logo 2 USB.zip](https://github.com/NeuronRobotics/nrjavaserial/files/6658702/NR.logo.2.USB.zip)

And I also created a new logo based on the USB symbol:

**Logo 3**
Font: Roboto Condensed (from [Google Fonts](https://fonts.google.com/specimen/Roboto+Condensed))
License: Creative Commons CC0 (Public Domain)

PNG version:
[image]

SVG version (inside the ZIP):
[NR logo 3.zip](https://github.com/NeuronRobotics/nrjavaserial/files/6658741/NR.logo.3.zip)

I don't know if the image resolution allows you to see clearly, but the logo 3 is inspired by the USB symbol format. However, in place of geometric shapes there are 3 types of connectors: on the left a DB9 connector (older), in the center a USB A connector (most common) and on the right a micro USB or USB C connector (newer). Also, since this symbol resembles a fork, you can think of an allusion to the fact that the NR Java Serial is an RXTX fork.

fwolter, 2021-06-15:
This is pretty cool! I really like the idea and the realization of the third logo!

MrDOS, 2021-06-16:
I liked logo 1, but logo 3 is dope as heck. I love the play on the USB logo.

My understanding is that the name of the project is NRJavaSerial, though, not NR Java Serial (no spaces). Can you try logo 3 without the spaces, and vertically align the text with the centre of the emblem (instead of its bottom)?

celsou, 2021-06-22:
> I liked logo 1, but logo 3 is dope as heck. I love the play on the USB logo.
>
> My understanding is that the name of the project is NRJavaSerial, though, not NR Java Serial (no spaces). Can you try [...]

Here is a version with no space between the words:
[image]

SVG image (inside ZIP): [NR logo 3.zip](https://github.com/NeuronRobotics/nrjavaserial/files/6695356/NR.logo.3.zip)

You can also change other aspects of the logo (color, alignment...) by editing the SVG file with Inkscape. You only need to install the Roboto Condensed font so that Inkscape does not use a fallback font.