NeuronRobotics/nrjavaserial #173: Create instructions for FreeBSD to cross-compile on an Ubuntu target

madhephaestus, 2020-05-06:
The intent is to get all of the cross-compileing to take place in CI. It would be nice for the CI to take care of the task of cross compiling all of the toolchains and placing binaries in the staging branch automatically.

RafalLukawiecki, 2020-05-06:
Adding your original instructions: https://marcelog.github.io/articles/cross_freebsd_compiler_in_linux.html

If the FreeBSD cross-compiler were to be set-up, would it need its own repository, or would it live inside this one? Do you use an external environment for the build process, if so, where is it?

madhephaestus, 2020-05-06:
the goal would be to make a shell script that gets called by the Freebsd make target that auto-downloads the tool and sets it up. Then it calls the compiler out of the folder it just created. I usually do this all with apt commands to install pre-packaged tools for the arm compiles. This may need to be a bit more elaborate for the freebsd build.

MrDOS, 2020-06-16:
I took a look on Docker Hub and found two existing FreeBSD cross-compilation environments, but neither worked for this project: both of them had a broken `termios.h` header.

By following a couple different FreeBSD cross-compiler guides (primarily the one linked above) and referencing the two existing FreeBSD cross-compilation environments on Docker Hub (which also seem to be based on that same guide), I've built [my own Docker container](https://hub.docker.com/r/empterdose/freebsd-cross-build) which works well for our purpose. Based on my updated `.travis.yml` from #189:

```
$ docker pull empterdose/freebsd-cross-build:9.3
$ cd /path/to/nrjavaserial
$ export JAVA_HOME=/usr/lib/jvm/adoptopenjdk-8-hotspot-amd64 # or wherever
$ docker run --interactive \
             --tty \
             --detach \
             --rm \
             --user $(id --user):$(id --group) \
             --name build \
             --env JAVA_HOME=$JAVA_HOME \
             --volume $(pwd):/build \
             --volume $JAVA_HOME:$JAVA_HOME \
             --volume $JAVA_HOME/include/linux:$JAVA_HOME/include/freebsd \
             empterdose/freebsd-cross-build:9.3
$ docker exec build make -C /build/src/main/c clean-freebsd
$ docker exec build settarget i386-freebsd9 make -C /build/src/main/c -j freebsd32
$ docker exec build settarget x86_64-freebsd9 make -C /build/src/main/c -j freebsd64
```

Note that this will only work on my `refactor-native-builds` branch until that gets merged into `master`: besides the fact the current native makefile has no `clean-freebsd` target, its FreeBSD targets don't use the `CC` environment variable, so it's not possible to override which compiler they use to build/link.

MrDOS, 2021-01-21:
Resolved by #189.

closed 2021-01-21 by MrDOS