NeuronRobotics/nrjavaserial #177: Additional constructors for NRSerialPort

riedlse, 2020-05-09:
Since there seems to be a lot of activity these days....

I did a fork a long time ago and thought I had logged an issue to get some additional constructors as I needed to set parity. Small mod to just add the other options to NRSerialPort. Rather then fork and pull, can you just add this?

    private RXTXPort serial;
    private String port = null;
    private boolean connected = false;
    private int baud = 115200;
    private int parity = SerialPort.PARITY_NONE;
    private int dataBits = SerialPort.DATABITS_8;
    private int stopBits = SerialPort.STOPBITS_1;
    
    /**
     * Class Constructor for a NRSerialPort with a given port and baudrate.
     * 
     */
    public NRSerialPort(String port)
    {
        this.port = port;
    }

    public NRSerialPort(String port, int baud)
    {
        this.port = port;
        this.baud = baud;
    }
    
    public NRSerialPort(String port, int baud, int parity)
    {
        this.port = port;
        this.baud = baud;
        this.parity = parity;
    }

    public NRSerialPort(String port, int baud, int parity, int dataBits)
    {
        this.port = port;
        this.baud = baud;
        this.parity = parity;
        this.dataBits = dataBits;
    }

    public NRSerialPort(String port, int baud, int parity, int dataBits, int stopBits)
    {
        this.port = port;
        this.baud = baud;
        this.parity = parity;
        this.dataBits = dataBits;
        this.stopBits = stopBits;
    }

madhephaestus, 2020-05-09:
Please submit a PR with these changes :)

riedlse, 2020-05-09:
Think I did it did a pull request and merge correctly.... waiting on Travis.... I think....

riedlse, 2020-05-09:
Sorry, messed up, didn't notice the latest uses getters and setters, re forked and new pull request coming, if somehow the old one makes it, ummm?

closed 2020-05-09 by riedlse