NeuronRobotics/nrjavaserial #181: Disconnecting USB device causes Java to exit

robnielsen, 2020-05-22:
The following causes Java to exit with no error in the following situation on Linux 

1. disconnect a working USB device, this causes a IOException to get thrown
1. close the IO streams associated with the device
1. open the USB device

This causes Java to exit in gnu.io.CommPortIdentifier.open():

```
throw new gnu.io.PortInUseException(
	native_psmisc_report_owner(PortName));
```
With nrjavaserial-3.14.0.jar that is used with openHAB, this is line 474 in CommPortIdentifier.java.

It appears that Java is crashing while trying to call the native method `native_psmisc_report_owner`.

This appears to be the same as https://code.google.com/archive/p/nrjavaserial/issues/27

madhephaestus, 2020-05-22:
Disconnect events were added in 5.1.x, please try the current release before opening an issue.

robnielsen, 2020-05-22:
@madhephaestus it still is happening with 5.1.1. See https://github.com/openhab/openhab-core/issues/1492#issuecomment-632696621

madhephaestus, 2020-05-22:
it seems like there are a bunch on exit() calls in the fuserimp code. They seem to indicate serious errors in file allocation events. To make this stable the functions need to be changed to have a return code (they are all void at the moment) and the calling JNI needs to throw an exception.

robnielsen, 2020-05-22:
At this point the port shouldn't be in use either, both the read and write streams were closed.

madhephaestus, 2020-05-22:
well there is a much easier fix to this issue than changing all the native code, just throw the exception rather than diving into native code to lookup the owner . 

I will push out a release today

madhephaestus, 2020-05-22:
release 5.2.0 should resolve this issue

closed 2020-05-22 by madhephaestus