NeuronRobotics/nrjavaserial #183: Detect if RFC2217 server disappears/is closed

bodiroga, 2020-05-29:
Hi @madhephaestus!

I'm (with the help of @wborn) trying to improve the RFC2217 experience in openHAB and, while making some changes to the openHAB Z-Wave binding, I have realized that the nrjavaserial library is not able to detect when the RFC2217 connection is closed by the server. Well, to be honest, I'm able to detect that something goes wrong when trying to send a message through the "tunnel", but, ideally, the library should be able to information the application when the connection is closed, similarly to how the USB plug/unplug detection has been implemented.

As the RFC2217 server I have tried with two different implementation: the pyserial [rfc2217 example](https://github.com/pyserial/pyserial/blob/master/examples/rfc2217_server.py) and [ser2net](https://linux.die.net/man/8/ser2net), although both cases have shown the same result.

Would it be possible to add it as a SerialPortEvent?

Many thanks for your work on the library and it's great to see that the development has resumed :+1: 

Best regards,

Aitor

madhephaestus, 2020-05-30:
Funny you should ask, this wasn't implemented in the hardware serial ports until about a month ago. It does now exist as a feature for usb-serial adapters. This is actuated in the event loop by attempting to read the number of bytes availible, which throws an exception. 

As a temp workaround, can you get it to crash with a polling read number of bytes availible? That would be ideal since it doesn't involve data transfer.

bodiroga, 2020-05-30:
> Funny you should ask, this wasn't implemented in the hardware serial ports until about a month ago. It does now exist as a feature for usb-serial adapters. This is actuated in the event loop by attempting to read the number of bytes availible, which throws an exception.

Yeah, I know it, my issue was more a feature request than an actual issue, sorry if that wasn't clear. The recent changes done to the library have been awesome, thank you very much for your work @madhephaestus :wink: 

> As a temp workaround, can you get it to crash with a loop thread polling the read number of bytes availible? That would be ideal since it doesn't involve data transfer.

Ok, I think that I've found a workaround. When a rfc2217 port is used, the nrjavaserial library doesn't allow the ```enableReceiveTimeout()``` function, right? That is how openHAB behaves, but it is a ```nrjavaserial``` feature, right? What I have seen is that, when the RFC2217 server is closed, the ```read()``` function returns -1, and that's not excepted when the receiveTimeout is not enabled. So, that value can be checked and, consequently, close the rfc2217 client connection. I will implement this solution and report back how it goes :wink: 

Thanks for your help!

bodiroga, 2020-05-31:
I can confirm that checking if the ```read()``` function returns -1 when ```enableReceiveTimeout``` is not supported works really great :+1: