NeuronRobotics/nrjavaserial #186: Q: How to get OpenJDK ARM64 running

Gerry33, 2020-06-03:
Hi all, and thanks for this work.
I'm using a ARM64 system with 

> openjdk version "11.0.7" 2020-04-14 LTS
> OpenJDK Runtime Environment Zulu11.39+61-CA (build 11.0.7+10-LTS)
> OpenJDK 64-Bit Server VM Zulu11.39+61-CA (build 11.0.7+10-LTS, mixed mode)
> 

My usage of nrjavaserial crashes at startup with
[
Exception in thread "main" java.lang.NoClassDefFoundError: gnu/io/SerialPortEventListener
        at java.base/java.lang.Class.forName0(Native Method)
        at java.base/java.lang.Class.forName(Class.java:398)](url)

I read somewhere in the bugs that there is option for OpenJDK to run, but I cannot find any hint about how and where.
I would be very thanksfull for a hint.
Regards
Gery

MrDOS, 2020-06-03:
The library should work on OpenJDK 11 and ARM64 without any particular intervention on your part.

* What version of the library are you using?
* Can you share how you're including it in your project – Maven, Gradle, etc.?
* When do you get this error: right at application start, or only after you call `CommPortIdentifier.getPortIdentifiers()`?

Gerry33, 2020-06-03:
Hi and thanks for taking care.
> What version of the library are you using?
> Can you share how you're including it in your project – Maven, Gradle, etc.?
MAVEN:
```
<dependency>
    	<groupId>com.neuronrobotics</groupId>
    	<artifactId>nrjavaserial</artifactId>
    	**<version>5.1.1</version>**
    </dependency>

```
> When do you get this error: right at application start, ...
Right at application start. I have a sysout as the very first statement after main() which does not appear.

I start using a runnable jar. 
'nrjavaserial-5.1.1.jar' is contained in that jar at root level.

"Hello World" and DACAPO benchmarks are running, so likely no issue with JAVA at all. 
Anyway, I'll inspect my JAVA Installation again. 

-- Update: The OpenJDK I'm using, has a classloader issue. Another jar built by me,  has similar issues.
Anyway, any help is appreciated. 

Gerry

closed 2020-06-03 by madhephaestus